tinygo-org/tinygo #3886: Previously ok build (in 0.28.1) fails in 0.29.0 with "verification error after compiling" error

joeshaw, 2023-09-01:
If you check out https://github.com/fastly/compute-starter-kit-go-default at the current main ([0cc8e79](https://github.com/fastly/compute-starter-kit-go-default/commit/0cc8e7982027a276ee495ce86d313b28e54a97d8)) and compile it with `tinygo build -target=wasi -gc=conservative -o bin/main.wasm .` on TinyGo 0.29.0 with Go 1.20.7 (or 1.21) you'll get these errors:

```
Incorrect number of arguments passed to called function!
  %30 = call i32 @write(i32 %13, i32 %28, i32 %29, ptr %nWritten), !dbg !595
Incorrect number of arguments passed to called function!
  %16 = call i32 @lookup(i32 %8, i32 %14, i32 %15, ptr %13), !dbg !2334
Incorrect number of arguments passed to called function!
  %13 = call i32 @get(i32 %5, i32 %11, i32 %12, ptr %10), !dbg !2458
Call parameter type does not match function signature!
  %16 = extractvalue %"github.com/fastly/compute-sdk-go/internal/abi/prim.ArrayU8" %10, 0, !dbg !3130
 ptr  %18 = call i32 @lookup(i32 %16, i32 %17, i32 %13, ptr %14, ptr %15), !dbg !3130
Incorrect number of arguments passed to called function!
  %17 = call i32 @insert(i32 %15, i32 %16, i32 %12, ptr %13, ptr %14), !dbg !3156
error: verification error after compiling package github.com/fastly/compute-sdk-go/internal/abi/fastly
```

These errors weren't present in 0.28.1.

I've bisected the repo and it was commit d845f1e1b2f04b1c1e778bb7caac005967c1ed7c that first started causing these.

I haven't done more digging than this.  I am not sure if these errors are legitimate or not (though things seemed to be working fine with 0.28.1 and previous versions).  A newer version of the starter kit upgrades the Fastly SDK to 0.2.0, which moves to the new Go 1.21-compatible `go:wasmimport` scheme and that works fine, but we would like to maintain compatibility with older versions of the SDK if possible.

joeshaw, 2023-09-01:
That code is using both `//go:wasm-module` and `//export`, which d845f1e says is a no-no:

> `When a function is exported using //export, but also had a //go:wasm-module pragma, the //export name was ignored. The //go:wasm-module doesn't actually do anything besides breaking the export (exported functions don't have a module name).`

If you look in `hostcalls_guest.go` linked above, there are several functions with the same name in different wasm modules.  For instance, `lookup` appears in the `fastly_geo`, `fastly_object_store`, and `fastly_cache` modules.  It seems like specifying the module name in addition to the export name is necessary to distinguish between them.

joeshaw, 2023-09-01:
The other functions listed in the error message also have multiple definitions in different modules with different parameter lists, which would explain why they appear.  For instance:

`write` in `fastly_http_body`: https://github.com/fastly/compute-sdk-go/blob/v0.1.7/internal/abi/fastly/hostcalls_guest.go#L239-L257

`write` in `fastly_log`: https://github.com/fastly/compute-sdk-go/blob/v0.1.7/internal/abi/fastly/hostcalls_guest.go#L373-L390

Dunno about the `prim.ArrayU8` error yet though.

aykevl, 2023-09-02:
It appears that the `//export` pragmas that were previously ignored due to a bug, are now respected and as a result the code breaks. So the bug was always there, but surfaces only now because a bug in TinyGo prevented it from surfacing.

If that's the case, the easy fix would be to remove the `//export` declarations. Or at least rename them to something unique: the name has to be unique in the binary (not just a particular package).

deadprogram, 2024-12-12:
@joeshaw I'm pretty sure this was already resolved some versions ago, so can we close this issue?

closed 2024-12-12 by joeshaw