tinygo-org/tinygo #3888: Invalid read in runtime.startMark in valgrind

QuLogic, 2023-09-03:
When running on aarch64, but building for x86_64 and qemu 8.1.0, I'm getting a crash running the `reflect.go` test. This _doesn't_ happen with qemu 7 or a native build. In order to debug this, I tried running the produce executable in valgrind natively. This generates a few invalid read and uninitialised value errors. Now, I know that Golang is difficult to run in valgrind due to its GC, so I'm not sure how much this can be trusted, but I wonder about the invalid read? Is this a true problem that only crashes in qemu?
```
sizes:
int8 1 8
int16 2 16
int32 4 32
int64 8 64
uint8 1 8
uint16 2 16
uint32 4 32
uint64 8 64
float32 4 32
float64 8 64
complex64 8 64
complex128 16 128
offset for int64 matches: true
offset for complex128 matches: true
==1831880== Invalid read of size 8
==1831880==    at 0x20A42F: runtime.startMark (gc_blocks.go:531)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==  Address 0x480f858 is in a rw- anonymous segment
==1831880==
==1831880== Conditional jump or move depends on uninitialised value(s)
==1831880==    at 0x20A488: (*runtime.gcObjectScanner).nextIsPointer (gc_precise.go:0)
==1831880==    by 0x20A488: runtime.startMark (gc_blocks.go:533)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==
==1831880== Use of uninitialised value of size 8
==1831880==    at 0x20A4CC: (runtime.gcBlock).state (gc_blocks.go:150)
==1831880==    by 0x20A4CC: runtime.startMark (gc_blocks.go:541)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==
==1831880== Conditional jump or move depends on uninitialised value(s)
==1831880==    at 0x20A4DA: runtime.startMark (gc_blocks.go:0)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==
==1831880== Use of uninitialised value of size 8
==1831880==    at 0x20A629: (runtime.gcBlock).state (gc_blocks.go:150)
==1831880==    by 0x20A629: (runtime.gcBlock).findHead (gc_blocks.go:124)
==1831880==    by 0x20A4EA: runtime.startMark (gc_blocks.go:551)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==
==1831880== Conditional jump or move depends on uninitialised value(s)
==1831880==    at 0x20A644: (runtime.gcBlock).findHead (gc_blocks.go:0)
==1831880==    by 0x20A4EA: runtime.startMark (gc_blocks.go:551)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==
==1831880== Use of uninitialised value of size 8
==1831880==    at 0x20A4F2: (runtime.gcBlock).state (gc_blocks.go:150)
==1831880==    by 0x20A4F2: runtime.startMark (gc_blocks.go:553)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==
==1831880== Conditional jump or move depends on uninitialised value(s)
==1831880==    at 0x20A508: runtime.startMark (gc_blocks.go:0)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==
==1831880== Conditional jump or move depends on uninitialised value(s)
==1831880==    at 0x20A47F: (*runtime.gcObjectScanner).nextIsPointer (gc_precise.go:0)
==1831880==    by 0x20A47F: runtime.startMark (gc_blocks.go:533)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==
==1831880== Use of uninitialised value of size 8
==1831880==    at 0x20A538: (runtime.gcBlock).setState (gc_blocks.go:158)
==1831880==    by 0x20A538: runtime.startMark (gc_blocks.go:562)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==
==1831880== Use of uninitialised value of size 8
==1831880==    at 0x20A343: runtime.newGCObjectScanner (gc_precise.go:74)
==1831880==    by 0x20A343: runtime.startMark (gc_blocks.go:517)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==
==1831880== Use of uninitialised value of size 8
==1831880==    at 0x20A009: runtime.sweep (gc_blocks.go:0)
==1831880==    by 0x20A009: runtime.runGC (gc_blocks.go:458)
==1831880==    by 0x20A009: runtime.alloc (gc_blocks.go:303)
==1831880==    by 0x20DEBB: runtime.sliceAppend (slice.go:30)
==1831880==    by 0x20D6F2: UnknownInlinedFun (reflect.go:507)
==1831880==    by 0x20D6F2: UnknownInlinedFun (reflect.go:315)
==1831880==    by 0x20D6F2: runtime.run$1 (scheduler_any.go:25)
==1831880==    by 0x20D6F2: runtime.run$1$gowrapper (scheduler_any.go:23)
==1831880==    by 0x2063F5: ??? (task_stack_amd64.S:29)
==1831880==
type assertion succeeded for unreferenced type

alignment / offset:
struct{[0]func(); byte}: true

struct tags
blue gopher

v.Interface() method
kind: interface
int 5
```

aykevl, 2023-09-03:
I tried to run a simple binary under Valgrind (linux/arm64, not amd64) and got the same "Conditional jump or move depends on uninitialised value(s)" error. I'm not sure why it's happening, but I'd guess Valgrind doesn't understand how TinyGo allocates memory. The fact that the tests of lots of real-world packages does in fact work while Valgrind immediately flags a supposed memory error suggests to me that this is in fact a false positive.

That said, I can reproduce the crash in Qemu. But if this happens only in Qemu and not natively (as you say), I'm inclined to think it's actually a Qemu bug, not a TinyGo bug.

(I wouldn't be surprised if it was a Qemu bug. Qemu can't seem to run _any_ i386 binaries built by TinyGo so I don't have a lot of trust in the x86_64 variant of it).

QuLogic, 2023-09-04:
> I tried to run a simple binary under Valgrind (linux/arm64, not amd64) and got the same "Conditional jump or move depends on uninitialised value(s)" error. I'm not sure why it's happening, but I'd [...]

I expect it's similar to the regular Go memory handling. I'm not too worried with the uninitialised memory, but I was concerned about the invalid read:
```
==1831880== Invalid read of size 8
==1831880==    at 0x20A42F: runtime.startMark (gc_blocks.go:531)
==1831880==    by 0x2063E1: ??? (asm_amd64.S:21)
==1831880==    by 0xC01FFFFFFFFFFFFF: ???
==1831880==  Address 0x480f858 is in a rw- anonymous segment
```
Perhaps the anonymous segment part is what's tripping up QEMU?

> That said, I can reproduce the crash in Qemu. But if this happens only in Qemu and not natively (as you say), I'm inclined to think it's actually a Qemu bug, not a TinyGo bug.

Unfortunately, all I could get out of QEMU was a backtrace through it's normal signal handler, which didn't seem to explain anything to me.

> (I wouldn't be surprised if it was a Qemu bug. Qemu can't seem to run _any_ i386 binaries built by TinyGo so I don't have a lot of trust in the x86_64 variant of it).

That's probably https://gitlab.com/qemu-project/qemu/-/issues/447 and https://gitlab.com/qemu-project/qemu/-/issues/690, but x86_64 always seemed to work.

aykevl, 2023-09-04:
> I'm not too worried with the uninitialised memory, but I was concerned about the invalid read:
>
> ```
> ==1831880== Invalid read of size 8
> ==1831880==    at 0x20A42F: runtime.startMark [...]

TinyGo uses `mmap` to allocate a large chunk of memory on startup, I think that's what QEMU refers to as "anonymous segment" (it's an anonymous mapping, not a file mapping).
I'm not sure why Valgrind is complaining there: it runs just fine outside Valgrind and I'd expect at least a segfault if there was a true invalid read.

> Unfortunately, all I could get out of QEMU was a backtrace through its normal signal handler, which didn't seem to explain anything to me.

I can maybe try to get gdb working with the emulated process, but my main laptop uses Arch Linux which doesn't even include gdb-multiarch. Would be interesting to see where it crashes exactly.

> That's probably https://gitlab.com/qemu-project/qemu/-/issues/447 and https://gitlab.com/qemu-project/qemu/-/issues/690, but x86_64 always seemed to work.

It's neither. On i386, it just crashes with a segmentation fault and not with a real error message. I reported it upstream here: https://gitlab.com/qemu-project/qemu/-/issues/1267 (they say it's the lack of vDSO support).

deadprogram, 2024-12-06:
Is this still an issue or can it be closed? @QuLogic @aykevl

aykevl, 2024-12-06:
QEMU used to be quite buggy for me, but nowadays (with Fedora Asahi 40) I don't have any issues anymore.

I'm going to close this, it can be reopened if it still occurs.

closed 2024-12-06 by aykevl