tinygo-org/tinygo #3890: Faulty WASM During Initialization

cryi, 2023-09-03:
## Environment:
- tinygo version: 0.29.0
- operating System: windows and wsl2 (ubuntu 22.04)
- architecture: WASM

## Description:
Compiling the given code for WASM using tinygo  produces incorrect behavior. The function checksum() returns a value that doesn't match the expected one only when it is called during the initialization of global variables. The issue appears to be related to the function panicOnError() that is declared in the code but is not directly used for the failing part.

## Minimal Reproducible Example:
```go
//go:build js && wasm

package main

import (
	"crypto/sha256"
	"errors"
	"fmt"
)

var (
	ProtoAlpha = MustParseProtocolHash("ProtoALphaALphaALphaALphaALphaALphaALphaALphaDdp3zK")
	input      = []byte{2, 170, 11, 205, 123, 44, 173, 205, 135, 236, 176, 213, 197, 3, 48, 251, 89, 254, 237, 116, 50, 191, 254, 206, 222, 138, 9, 162, 184, 109, 21, 39, 198, 165}
)

func checksum(input []byte) (cksum [4]byte) {
	h := sha256.Sum256(input)
	h2 := sha256.Sum256(h[:])
	copy(cksum[:], h2[:4])
	return
}

func test() error {
	expected := [4]byte{220, 247, 41, 156}
	if checksum(input) /* [220 247 41 156]*/ != expected {
		return errors.New("checksum mismatch")
	}
	return nil
}

func MustParseProtocolHash(src string) (h [32]byte) {
	err := test()
	panicOnError(err)
	// uncomment this and code works
	// if err != nil {
	// 	panic(err)
	// }
	return h
}

func panicOnError(err error) {
	if err != nil {
		panic(err)
	}
}

func main() {
	// trigger init of tezos module
	fmt.Println(ProtoAlpha)

	// this works too
	// fmt.Println(MustParseProtocolHash("ProtoALphaALphaALphaALphaALphaALphaALphaALphaDdp3zK"))
}
```

## Steps to reproduce:
1. compile the provided code using tinygo targeting WASM. (`tinygo clean && tinygo build -o main.wasm -target wasm -no-debug`)
2. run the resulting WASM binary. 
3. observe that the checksum generated during global variable initialization doesn't match the expected value.

## Expected behavior:

The checksum generated should match the expected value `[220 247 41 156]`.

## Actual behavior:

The checksum doesn't match the expected value when computed during the initialization of global variables. Seems like memory corruption.

## Notes:

When the function panicOnError() is replaced by inline version or if the checksum computation occurs after global variable initialization, the issue does not manifest.

Any help on this would be greatly appreciated!

aykevl, 2023-09-04:
I have a fix (it was indeed `copy`), now I need to write tests etc.

cryi, 2023-09-04:
Sure, I will try to test today/tomorrow and let you know. Tank you 👍🏻

cryi, 2023-09-06:
Ok, this fixed the initial loading issue. Unfortunately now I face stack overflow caused by GC and I am not able to tell with certainty whether it may be related to initialization phase or not. I tried to narrow down the code but so far all efforts failed. 😞

cryi, 2023-09-06:
Ok now @dgryski pointed me to `-stack-size=128KB` from #3258. It solves the stack overflow issue for me. So this fix indeed works as expected. Thank you @aykevl.

aykevl, 2023-09-09:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/3901

deadprogram, 2023-09-21:
This is addressed as part of `v0.30.0` release so now closing. Thanks!

closed 2023-09-21 by deadprogram