tinygo-org/tinygo #3891: Bootloader Implementation and Setting Firmware Start Address in RP2040 with TinyGo

lePereT, 2023-09-03:
Hi TinyGophers!

I'm working with the RP2040 and looking into using TinyGo for bootloader development. My goals:

  * The bootloader should manage firmware updates, determining which firmware partition to boot from based on flags in flash memory.
  * Due to the ARM Cortex-M0's lack of an MMU, the firmware needs to be compiled to run from its specific flash location.

Questions:

  1. How can I configure TinyGo to ensure the firmware is compiled to run from its precise location in flash, considering there's a bootloader and an unformatted space before the main firmware?
  2. Can TinyGo jump to and execute code at another flash location (like booting from a bootloader)? Are there built-in methods or accepted approaches, or is inline assembly the recommended approach?

Appreciated

Thank you!

lePereT, 2023-09-13:
Can someone help me out with this? Even just a pointer to some source code would be appreciated

aykevl, 2023-09-19:
> How can I configure TinyGo to ensure the main firmware is compiled to run from its precise location in flash, considering there will be an extra bootloader and space (for environment variables) before the main firmware?

For this, you'd need either your own linker script, or sometimes you can hook into the existing linker script to change the flash offsets. See for example how this is done for the Nordic SoftDevice: https://github.com/tinygo-org/tinygo/blob/release/targets/nrf52-s132v6.json

> Can TinyGo jump to and execute code at another flash location (like booting from a bootloader)? Are there built-in methods or accepted approaches, or is inline assembly the recommended approach?

I recommend writing inline assembly in C, and calling it using CGo. (Or if plain C is good enough, you don't even need inline assembly).
Don't worry about CGo overhead, it doesn't exist in TinyGo and most of the other disadvantages of CGo (cross compiler toolchains etc) don't really exist either.

lePereT, 2023-09-19:
Thank you Ayke, this gives me enough to get started with some confidence. I'll close this now, and also say thanks to you and all the TinyGo team for this important and ambitious project.

closed 2023-09-19 by lePereT