tinygo-org/tinygo #3892: Legacy passes removed in LLVM17

QuLogic, 2023-09-04:
It looks like pretty much all [the legacy pass API has been removed in LLVM17](https://github.com/llvm/llvm-project/blob/release/17.x/llvm/docs/ReleaseNotes.rst#changes-to-the-c-api). This covers everything from `go-llvm/transforms_*.go` as well as `LLVMConstSelect`. The latter is not used by tinygo, but several of the former are:

<details>
<summary>Functions from transforms_ipo.go</summary>

```
$ rg func lib/go-llvm/transforms_ipo.go
24:func boolToUnsigned(b bool) C.unsigned {
31:func (pm PassManager) AddConstantMergePass()      { C.LLVMAddConstantMergePass(pm.C) }
32:func (pm PassManager) AddDeadArgEliminationPass() { C.LLVMAddDeadArgEliminationPass(pm.C) }
33:func (pm PassManager) AddFunctionAttrsPass()      { C.LLVMAddFunctionAttrsPass(pm.C) }
34:func (pm PassManager) AddFunctionInliningPass()   { C.LLVMAddFunctionInliningPass(pm.C) }
35:func (pm PassManager) AddGlobalDCEPass()          { C.LLVMAddGlobalDCEPass(pm.C) }
36:func (pm PassManager) AddGlobalOptimizerPass()    { C.LLVMAddGlobalOptimizerPass(pm.C) }
37:func (pm PassManager) AddIPSCCPPass()             { C.LLVMAddIPSCCPPass(pm.C) }
38:func (pm PassManager) AddInternalizePass(allButMain bool) {
41:func (pm PassManager) AddStripDeadPrototypesPass() { C.LLVMAddStripDeadPrototypesPass(pm.C) }

$ rg 'AddConstantMergePass|AddDeadArgEliminationPass|AddFunctionAttrsPass|AddFunctionInliningPass|AddGlobalDCEPass|AddGlobalOptimizerPass|AddIPSCCPPass|AddInternalizePass|AddStripDeadPrototypesPass' -g'!lib'
transform/interface-lowering_test.go
20:             pm.AddGlobalDCEPass()

transform/optimizer.go
91:             goPasses.AddGlobalDCEPass()
92:             goPasses.AddGlobalOptimizerPass()
93:             goPasses.AddIPSCCPPass()
96:             goPasses.AddFunctionAttrsPass()
139:            goPasses.AddGlobalDCEPass()

transform/allocs_test.go
44:     pm.AddFunctionAttrsPass()

interp/interp_test.go
84:     pm.AddGlobalOptimizerPass()
```
</details>

<details>
<summary>Functions from transforms_pmbuilder.go</summary>

```
$ rg func lib/go-llvm/transforms_pmbuilder.go
27:func NewPassManagerBuilder() (pmb PassManagerBuilder) {
32:func (pmb PassManagerBuilder) SetOptLevel(level int) {
36:func (pmb PassManagerBuilder) SetSizeLevel(level int) {
40:func (pmb PassManagerBuilder) Populate(pm PassManager) {
44:func (pmb PassManagerBuilder) PopulateFunc(pm PassManager) {
48:func (pmb PassManagerBuilder) Dispose() {
52:func (pmb PassManagerBuilder) SetDisableUnitAtATime(val bool) {
56:func (pmb PassManagerBuilder) SetDisableUnrollLoops(val bool) {
60:func (pmb PassManagerBuilder) SetDisableSimplifyLibCalls(val bool) {
64:func (pmb PassManagerBuilder) UseInlinerWithThreshold(threshold uint) {

$ rg NewPassManagerBuilder -g'!lib'
transform/optimizer.go
25:     builder := llvm.NewPassManagerBuilder()
58:     builder := llvm.NewPassManagerBuilder()
```
(I didn't bother looking for any of the builder methods since the struct itself is removed anyway.)
</details>

<details>
<summary>Functions from transforms_scalar.go</summary>

```
rg func lib/go-llvm/transforms_scalar.go
24:func (pm PassManager) AddAggressiveDCEPass()           { C.LLVMAddAggressiveDCEPass(pm.C) }
25:func (pm PassManager) AddCFGSimplificationPass()       { C.LLVMAddCFGSimplificationPass(pm.C) }
26:func (pm PassManager) AddDeadStoreEliminationPass()    { C.LLVMAddDeadStoreEliminationPass(pm.C) }
27:func (pm PassManager) AddGVNPass()                     { C.LLVMAddGVNPass(pm.C) }
28:func (pm PassManager) AddIndVarSimplifyPass()          { C.LLVMAddIndVarSimplifyPass(pm.C) }
29:func (pm PassManager) AddInstructionCombiningPass()    { C.LLVMAddInstructionCombiningPass(pm.C) }
30:func (pm PassManager) AddJumpThreadingPass()           { C.LLVMAddJumpThreadingPass(pm.C) }
31:func (pm PassManager) AddLICMPass()                    { C.LLVMAddLICMPass(pm.C) }
32:func (pm PassManager) AddLoopDeletionPass()            { C.LLVMAddLoopDeletionPass(pm.C) }
33:func (pm PassManager) AddLoopRotatePass()              { C.LLVMAddLoopRotatePass(pm.C) }
34:func (pm PassManager) AddLoopUnrollPass()              { C.LLVMAddLoopUnrollPass(pm.C) }
35:func (pm PassManager) AddMemCpyOptPass()               { C.LLVMAddMemCpyOptPass(pm.C) }
36:func (pm PassManager) AddPromoteMemoryToRegisterPass() { C.LLVMAddPromoteMemoryToRegisterPass(pm.C) }
37:func (pm PassManager) AddReassociatePass()             { C.LLVMAddReassociatePass(pm.C) }
38:func (pm PassManager) AddSCCPPass()                    { C.LLVMAddSCCPPass(pm.C) }
39:func (pm PassManager) AddScalarReplAggregatesPass()    { C.LLVMAddScalarReplAggregatesPass(pm.C) }
40:func (pm PassManager) AddScalarReplAggregatesPassWithThreshold(threshold int) {
43:func (pm PassManager) AddSimplifyLibCallsPass()       { C.LLVMAddSimplifyLibCallsPass(pm.C) }
44:func (pm PassManager) AddTailCallEliminationPass()    { C.LLVMAddTailCallEliminationPass(pm.C) }
45:func (pm PassManager) AddDemoteMemoryToRegisterPass() { C.LLVMAddDemoteMemoryToRegisterPass(pm.C) }
46:func (pm PassManager) AddVerifierPass()               { C.LLVMAddVerifierPass(pm.C) }

$ rg 'AddAggressiveDCEPass|AddCFGSimplificationPass|AddDeadStoreEliminationPass|AddGVNPass|AddIndVarSimplifyPass|AddInstructionCombiningPass|AddJumpThreadingPass|AddLICMPass|AddLoopDeletionPass|AddLoopRotatePass|AddLoopUnrollPass|AddMemCpyOptPass|AddPromoteMemoryToRegisterPass|AddReassociatePass|AddSCCPPass|AddScalarReplAggregatesPass|AddScalarReplAggregatesPassWithThreshold|AddSimplifyLibCallsPass|AddTailCallEliminationPass|AddDemoteMemoryToRegisterPass|AddVerifierPass' -g'!lib'
transform/optimizer.go
94:             goPasses.AddInstructionCombiningPass() // necessary for OptimizeReflectImplements
95:             goPasses.AddAggressiveDCEPass()

transform/allocs_test.go
43:     pm.AddInstructionCombiningPass()

transform/maps_test.go
20:             pm.AddDeadStoreEliminationPass()
21:             pm.AddAggressiveDCEPass()

interp/interp_test.go
85:     pm.AddDeadStoreEliminationPass()
86:     pm.AddAggressiveDCEPass()

compiler/compiler_test.go
96:                     funcPasses.AddInstructionCombiningPass()
```
</details>

QuLogic, 2023-09-04:
Ah, I thought I saw something about this before; looks like https://github.com/tinygo-org/go-llvm/pull/50 will be needed for this.

deadprogram, 2024-02-28:
Completed with `v0.31.0` so now closing. Thank you!

closed 2024-02-28 by deadprogram