tinygo-org/tinygo #3893: crash on interp timeout

dkegel-fastly, 2023-09-06:
I can reproduce this on both mac and linux.   I boiled it down on mac to this:

main.go:
```
package main
func main() {
	println("hello, world")
}
```

go.mod:
```
module main

go 1.21.0
```

To reproduce:
```
$ go mod tidy
$ tinygo build -interp-timeout=0s
  
# main
/Users/dkegel/tobug: interp: running for more than 0s, timing out (executed calls: 1)
  SIGSEGV: segmentation violation
PC=0x10423f89c m=11 sigcode=1
signal arrived during cgo execution

goroutine 1 [syscall]:
runtime.cgocall(0x1003e2960, 0xc0006e9460)
	/Users/runner/hostedtoolcache/go/1.21.0/x64/src/runtime/cgocall.go:157 +0x4b fp=0xc0006e9438 sp=0xc0006e9400 pc=0x10000934b
tinygo.org/x/go-llvm._Cfunc_LLVMDumpValue(0x600001718180)
	_cgo_gotypes.go:4906 +0x3f fp=0xc0006e9460 sp=0xc0006e9438 pc=0x10024c83f
main.printCompilerError.Value.Dump.func1({0xc000120400?})
	/Users/runner/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20221028183034-8341240c0b32/ir.go:722 +0x39 fp=0xc0006e9498 sp=0xc0006e9460 pc=0x1003b6579
tinygo.org/x/go-llvm.Value.Dump(...)
	/Users/runner/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20221028183034-8341240c0b32/ir.go:722
main.printCompilerError(0x104680bc0, {0x1046f2840?, 0xc00018c000?})
	/Users/runner/work/tinygo/tinygo/main.go:1293 +0x1d6 fp=0xc0006e9748 sp=0xc0006e9498 pc=0x1003b5a96
main.handleCompilerError({0x1046f2840?, 0xc00018c000?})
	/Users/runner/work/tinygo/tinygo/main.go:1326 +0x2f fp=0xc0006e9770 sp=0xc0006e9748 pc=0x1003b65cf
main.main()
...

deadprogram, 2023-09-11:
Seems like `module main` in the go.mod file is perhaps not valid?

https://github.com/golang/go/issues/28514#issuecomment-457734995

kenbell, 2023-09-11:
From walking the code, I think what is happening is...

- When the interp timeout happens, some LLVM datastructures are captured in the error
- When build.Build exits, it's cleaning up the LLVM datastructures, but passing out an error that still references them
- When the error is finally looked at to generate the relevant failure message, it walks off into unallocated memory (that's probably been re-used)

I guess the fix is to not allow LLVM datastructures to be passed in an error out of build.Build

kenbell, 2023-09-11:
See #3906

deadprogram, 2023-09-13:
@kenbell @dkegel-fastly what I was saying is that having a `go.mod file` with

`module main`

is actually invalid. It was working previously but was actually incorrect.

Change the name of the module in the `go.mod` file to a valid name and it should work as expected.

dgryski, 2023-11-04:
Is this fixed now?

dkegel-fastly, 2023-11-04:
Yes, mostly.  The error message still has a tiny traceback in it, but no longer seems to segfault: 
```
$ tinygo version
tinygo version 0.30.0 darwin/amd64 (using go version go1.21.0 and LLVM version 16.0.1)
$ go version
go version go1.21.0 darwin/amd64
$ go mod tidy
$ tinygo build -interp-timeout=0s
# runtime/cgo
/usr/local/tinygo/src/runtime/cgo: interp: running for more than 0s, timing out (executed calls: 1)
ret 

traceback:
/usr/local/tinygo/src/runtime/cgo:
ret 
```

It shouldn't print out that vestigal traceback, but then again, it's not scary enough to bother users, I think.

deadprogram, 2024-05-01:
Closing since this appears to no longer be an issue. Please reopen if needed. Thanks!

closed 2024-05-01 by deadprogram