tinygo-org/tinygo #3897: Compiler: disable checks for a scheduler

stoader, 2023-09-08:
Is there a way to tell the compiler to skip verifying the use of go routines when the '-scheduler=none' build option is used?

There are libraries which may use multiple go routines or none based on runtime configuration. In case such a library is used with a configuration which will result in not launching go routines we would use the '-scheduler=none' build option. However, this currently causes the build to fail with `attempted to start a go routine without a scheduler`

As an example this go routine launch https://github.com/klauspost/compress/blob/master/zstd/decoder.go#L236C14-L236C14 will never execute if this library is configured to use only a single go routine: https://github.com/klauspost/compress/blob/master/zstd/decoder.go#L236C14-L236C14

aykevl, 2023-09-08:
> Is there a way to tell the compiler to skip verifying the use of go routines when the `-scheduler=none` build option is used?

Not right now.

What would you like to see instead of an error? We can't just ignore it entirely. One option would be to convert the compile time error into a runtime error. Would that work for you?

stoader, 2023-09-08:
@aykevl a runtime error make sense as that would catch actual go routine starts.