tinygo-org/tinygo #3907: Bug: syscall.ERANGE is not defined, cannot compile

Orphoros, 2023-09-12:
# Environment:

tinygo version: 0.29.0
operating System: macOS
architecture: aarch64

# Description:

I have a bigger Go project that I am trying to compile with Tinygo. Compilation brakes with an undefined syscall: `syscall.ERANGE`

# Steps to reproduce:

Given the following Go-code:

```go
package main

import (
	"syscall"
)

func main() {
	flags := syscall.ERANGE
	println(flags)
}
```

Run 

```
tinygo build main.go
```

# Expected behavior:

The app should print `34` to the terminal.

# Actual behavior:

Compilation brakes with the following error message:

```
# command-line-arguments
main.go:8:19: undefined: syscall.ERANGE
```

dkegel-fastly, 2023-09-12:
Which system calls do you expect to use that value?  Knowing that might help.

Orphoros, 2023-09-12:
I am not entirely sure myself tbh. I wanted to try out Tinygo on this project of mine, which is rather big: https://github.com/Orphoros/Shark

And when I ran `tinygo build ./cmd/bin`, it gave this error:
```
# os/user
/opt/homebrew/Cellar/go/1.21.1/libexec/src/os/user/cgo_lookup_unix.go:177:30: undefined: syscall.ERANGE
```
I tried to run Tinygo against this project without any 3rd party library imports, and it still gave this error. I am unsure what triggered this syscall, as I do not run any manually.

dkegel-fastly, 2023-09-12:
Try building with -tags="purego noasm", often that makes dependencies more tinygo-friendly

Orphoros, 2023-09-12:
Thanks for the suggestion! I tried it, but it still results in the same error message. This is the go lib function (cgo_lookup_unix.go:177:30) where it breaks:

```go
// retryWithBuffer repeatedly calls f(), increasing the size of the
// buffer each time, until f succeeds, fails with a non-ERANGE error,
// or the buffer exceeds a reasonable limit.
func retryWithBuffer(startSize bufferKind, f func([]byte) syscall.Errno) error {
	buf := make([]byte, startSize)
	for {
		errno := f(buf)
		if errno == 0 {
			return nil
		} else if runtime.GOOS == "aix" && errno+1 == 0 {
			// On AIX getpwuid_r appears to return -1,
			// not ERANGE, on buffer overflow.
		} else if errno != syscall.ERANGE {
			return errno
		}
		newSize := len(buf) * 2
		if !isSizeReasonable(int64(newSize)) {
			return fmt.Errorf("internal buffer exceeds %d bytes", maxBufferSize)
		}
		buf = make([]byte, newSize)
	}
}
```

dgryski, 2023-09-13:
`os/user` needs `-tags=osusergo` to build with tinygo.

Orphoros, 2023-09-14:
I tried it with the `-tags=osusergo` tag. It still throws syscall errors, but now only on a dependency package. If I remove it, it compiles. When I try to run the compiled code, I get this error:

```
checkId: 22 should be 23
panic: bootstrap type wrong id: mapType mapType not <nil>
```

Also, thanks a lot for your help so far!

aykevl, 2024-02-19:
With https://github.com/tinygo-org/tinygo/pull/4056, we've replaced the os/user package entirely so this specific issue shouldn't happen anymore. (It might still fail to compile because of other reasons though).

deadprogram, 2024-02-28:
Completed with `v0.31.0` so now closing. Thank you!

closed 2024-02-28 by deadprogram