tinygo-org/tinygo #3910: Update wasi-libc

anuraaga, 2023-09-15:
I noticed wasi-libc is currently from the wasi-sdk 16 release and the latest wasi-sdk is now 20. Would it make sense to update it? I was notably missing the `getpagesize` function that was added in wasi-sdk 17.

deadprogram, 2023-09-16:
Seems like a very good idea to me!

deadprogram, 2023-09-17:
See #3661

aykevl, 2023-09-19:
Why do you need `getpagesize`? It's fixed for WebAssembly (at 64kB).

deadprogram, 2023-09-19:
Possibly for a custom runtime?

anuraaga, 2023-09-19:
I personally need it to use bdwgc with TinyGo, so just need it linkable but doesn't need to be used by TinyGo itself. Interestingly, bdwg had a very similar discussion to the wasi-libc PR.

https://github.com/ivmai/bdwgc/pull/534

I somewhat expect TinyGo will stick to the constant since this codebase understands Wasm closely, but other linked-in native libraries also reasonably may keep to being more generic. That seems to be why wasi-libc added it

https://github.com/WebAssembly/wasi-libc/commit/e066a8b9d9c15e4c67fbcaeed3b62470f8785ce0

aykevl, 2024-02-25:
wasi-libc was updated, so closing.

closed 2024-02-25 by aykevl