tinygo-org/tinygo #3912: Undefined symbols for __sync... when building for ARMv5 Linux targets

MDr164, 2023-09-18:
Currently when building for ARMv5 Linux, using flags `GOARCH=arm GOARM=5` the TinyGo compiler (0.29.0) returns following error:
```
tinygo:ld.lld: error: undefined symbol: __sync_val_compare_and_swap_4
>>> referenced by doc.go:166 (/usr/lib/go/src/sync/atomic/doc.go:166)
>>>               /home/user/.cache/tinygo/thinlto/llvmcache-C44B4F4EEE0E036D40F53EC1B801A495DE509B56:((*sync/atomic.Value).Store)
>>> referenced by doc.go:114 (/usr/lib/go/src/sync/atomic/doc.go:114)
>>>               /home/user/.cache/tinygo/thinlto/llvmcache-C44B4F4EEE0E036D40F53EC1B801A495DE509B56:((*sync/atomic.Value).Store)

tinygo:ld.lld: error: undefined symbol: __sync_lock_test_and_set_4
>>> referenced by doc.go:192 (/usr/lib/go/src/sync/atomic/doc.go:192)
>>>               /home/user/.cache/tinygo/thinlto/llvmcache-C44B4F4EEE0E036D40F53EC1B801A495DE509B56:((*sync/atomic.Value).Store)
>>> referenced by value.go:0 (/usr/lib/go/src/sync/atomic/value.go:0)
>>>               /home/user/.cache/tinygo/thinlto/llvmcache-C44B4F4EEE0E036D40F53EC1B801A495DE509B56:((*sync/atomic.Value).Store)
>>> referenced by scheduler_any.go:24 (/usr/lib/tinygo/src/runtime/scheduler_any.go:24)
>>>               /home/user/.cache/tinygo/thinlto/llvmcache-C44B4F4EEE0E036D40F53EC1B801A495DE509B56:(runtime.run$1$gowrapper)
>>> referenced 2 more times
failed to run tool: ld.lld
error: failed to link /tmp/tinygo1270461697/main: exit status 1
```

This behavior is not being observed for `GOARM=6` and `GOARM=7`

laurento, 2024-02-25:
It seems it is still the case using go v1.21.7 and tinygo v0.30.0

aykevl, 2024-02-25:
I looked into it a bit and the usual way these are implemented seems to be by calling into a kernel-provided function. That's somewhat involved, though it could be done.

What's the target where you need `GOARM=5`? It might help to give a solution.

laurento, 2024-02-25:
Before reading this issue, I was using `GOARM=5` to target my Raspberry Pi first generation. However, I then realized it is actually an `armv6l` so I started using `GOARM=6` and, as mentioned already, it works just fine.

MDr164, 2024-02-26:
The target for me is the Aspeed AST2500 SoC which is `armv6k` but without FPU. Since Go 1.22 we actually got softfloat for ARM merged so I'm currently building with `GOARM=6,softfloat`. Actually I'd have to check again with the softfloat variant if it builds. Thanks for reminding me of this issue.

deadprogram, 2024-05-01:
Hello all. Trying to determine if there is some specific things to be done on this issue, or it should be closed since I think this is actually covered by #4177 

What does everyone think?

MDr164, 2024-05-01:
The PR i put up solves the initial issue I had for me at least as I don't have to target v5 anymore with my v6 and v7 softfloat targets. Not sure if the root cause is fixed but for my usecase I'm fine with closing.

deadprogram, 2024-05-05:
Thanks @MDr164

closed 2024-05-05 by deadprogram