tinygo-org/tinygo #3916: Missing GOPROXY in TinyGo 0.29.0 docker image

henders, 2023-09-18:
It looks like the latest Go compilers store their default `go env` in the `$(go env GOROOT)/go.env` file which is missing on the `tinygo/tinygo:0.29.0` docker image. This leads to `go env` printing:

```
$ go env
GO111MODULE=''
GOARCH='arm64'
GOBIN=''
GOCACHE='/home/tinygo/.cache/go-build'
GOENV='/home/tinygo/.config/go/env'
GOEXE=''
GOEXPERIMENT=''
GOFLAGS=''
GOHOSTARCH='arm64'
GOHOSTOS='linux'
GOINSECURE=''
GOMODCACHE='/home/tinygo/go/pkg/mod'
GONOPROXY=''
GONOSUMDB=''
GOOS='linux'
GOPATH='/home/tinygo/go'
GOPRIVATE=''
GOPROXY=''
GOROOT='/usr/local/go'
GOSUMDB=''
GOTMPDIR=''
GOTOOLCHAIN=''
GOTOOLDIR='/usr/local/go/pkg/tool/linux_arm64'
GOVCS=''
GOVERSION='go1.21.0'
GCCGO='gccgo'
AR='ar'
CC='gcc'
CXX='g++'
CGO_ENABLED='1'
GOMOD='/dev/null'
GOWORK=''
CGO_CFLAGS='-O2 -g'
CGO_CPPFLAGS=''
CGO_CXXFLAGS='-O2 -g'
CGO_FFLAGS='-O2 -g'
CGO_LDFLAGS='-O2 -g'
PKG_CONFIG='pkg-config'
GOGCCFLAGS='-fPIC -pthread -Wl,--no-gc-sections -fmessage-length=0 -ffile-prefix-map=/tmp/go-build860323270=/tmp/go-build -gno-record-gcc-switches'
```

As you can see above `GOPROXY` is empty leading to compile errors:
```
$ tinygo build -o main.wasm -gc=custom -tags=custommalloc -scheduler=none -target=wasi ./cmd/main.go
go: downloading github.com/wasilibs/nottinygc v0.3.0
go: downloading github.com/tetratelabs/proxy-wasm-go-sdk v0.22.0
go: downloading github.com/alecholmes/xfccparser v0.1.0
go: downloading github.com/golang-jwt/jwt/v5 v5.0.0
internal/request_handler.go:4:2: GOPROXY list is not the empty string, but contains no entries
internal/jwt.go:4:2: GOPROXY list is not the empty string, but contains no entries
internal/config.go:10:2: GOPROXY list is not the empty string, but contains no entries
internal/plugin_entrypoint.go:5:2: GOPROXY list is not the empty string, but contains no entries
cmd/main.go:5:2: GOPROXY list is not the empty string, but contains no entries
```

To workaround this I have to execute  `go env -w  GOPROXY=https://proxy.golang.org,direct` or prefix the compile command with GOPROXY, e.g.: `GOPROXY=https://proxy.golang.org,direct tinygo build ....`

henders, 2023-09-18:
From looking the docker file, the `tinygo` image is based on `golang:1.21` which does have the `go.env` file.
```
$ docker run -it golang:1.21 bash

root@40273fb9e15c:/go# cat $(go env GOROOT)/go.env

# This file contains the initial defaults for go command configuration.
# Values set by 'go env -w' and written to the user's go/env file override these.
# The environment overrides everything else.

# Use the Go module mirror and checksum database by default.
# See https://proxy.golang.org for details.
GOPROXY=https://proxy.golang.org,direct
GOSUMDB=sum.golang.org

# Automatically download newer toolchains as directed by go.mod files.
# See https://go.dev/doc/toolchain for details.
GOTOOLCHAIN=auto
```

So somewhere that file is getting stripped along with other files in `/usr/local/go`.

deadprogram, 2023-09-19:
Marking to be closed on next release.

deadprogram, 2023-09-21:
This is part of `v0.30.0` release so now closing. Thanks!

closed 2023-09-21 by deadprogram