tinygo-org/tinygo #3926: compile failed with `undefined: C._Bool`

Patrick0308, 2023-09-22:
# Environment:
tinygo version: 0.30.0
operating System: macOS
architecture: aarch64

# Description:
I have a bigger Go project that I am trying to compile with Tinygo. Compilation brakes with an undefined syscall: syscall.ERANGE

# Steps to reproduce:
Given the following Go-code:
```
package main

/*
  #include "stdbool.h"
  bool test() {
     return true;
  }
*/
import "C"

import "fmt"

func main() {
	fmt.Printf("%s", C._Bool(C.test()))
}
```
Run

`tinygo run main.go`

# Expected behavior:
```
❯ tinygo run main.go
%!s(main._Ctype__Bool=true)%    
```

# Actual behavior:

```
❯ tinygo run main.go
# command-line-arguments
main.go:14:19: undefined: C._Bool
```

aykevl, 2023-09-22:
Well this led me down a rather deep rabbit hole to improve CGo compatibility: https://github.com/tinygo-org/tinygo/pull/3927

I'll look at the actual bug next.

aykevl, 2023-09-22:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/3928

Patrick0308, 2023-09-27:
#3928 is merged

closed 2023-09-27 by Patrick0308

deadprogram, 2023-10-04:
Reopening and tagging to close on next release.

Patrick0308, 2023-10-12:
I build dev branch on my own computer. I tested 
```
package main

/*
  #include "stdbool.h"
  bool test() {
     return true;
  }
*/
import "C"

import "fmt"

func main() {
	fmt.Printf("%s", C._Bool(C.test()))
}
```
It's ok.

However I tested https://github.com/fermyon/spin/pull/1794 example but building painc.
The error log is 
```
Building component tinygo-mysql with `/Users/patrick/projects/tinygo/build/tinygo build -target=wasi -gc=leaking -o main.wasm main.go`
# github.com/fermyon/spin/sdk/go/mysql
../../../tinygo/llvm-project/clang/lib/Headers/stdbool.h:20:14: undefined: C._Bool
Error: Build command for component tinygo-mysql failed with status Exited(1)
```

Is it another issue?

Patrick0308, 2023-10-13:
@aykevl Any idea？

aykevl, 2024-02-19:
@Patrick0308 my guess is that they're shipping their own TinyGo variant, that is still the latest release (and not the dev branch).