tinygo-org/tinygo #3930: Waweshare RP2040-Plus compatibility

user91840981230985410298374983498389, 2023-09-24:
The RP2040-Plus is simmilar to raspberry pi pico(same processor), but has more pins.
is it compatible when using target as rpi pico, if not, is it possible to make some workaround?
I'm new to go and microcontrollers, so sorry for any misunderstanding.

aykevl, 2023-09-24:
> is it compatible when using target as rpi pico

I don't know, I don't have the hardware. But I think there is a high chance it will work.

> if not, is it possible to make some workaround?

Yes, by adding full support for this board.

> if so, can i use the same target for all pico compatible/rp2040 boards? for eg. seeduino xiao rp2040

Why would you want to do this? At least the Xiao RP2040 is natively supported by TinyGo.
There are some differences between boards, but mostly they are differences in flash size, available pins, and maybe flash configuration.

user91840981230985410298374983498389, 2023-09-25:
> I don't know, I don't have the hardware. But I think there is a high chance it will work.
>
>
> Yes, by adding full support for this board.
>
>
> Why would you want to do this? At least the Xiao RP2040 is [...]

just wanted to know, because i have some quite uncommon board, and i wasn't sure if they would work,
thanks for the quick reply

closed 2023-09-25 by user91840981230985410298374983498389