tinygo-org/tinygo #3933: Homebrew install fails on M2 macbook

joelegasse, 2023-09-26:
I have a newer M2 macbook pro, and the homebrew version of the install does not seem to work. I was able to successfully follow the source installation steps to get things working, though.

```
% tinygo version
/opt/homebrew/bin/tinygo: line 2: /opt/homebrew/Cellar/tinygo/0.30.0/libexec/tinygo: Bad CPU type in executable
/opt/homebrew/bin/tinygo: line 2: /opt/homebrew/Cellar/tinygo/0.30.0/libexec/tinygo: Undefined error: 0
```

```
% uname -a
Darwin macbook 22.6.0 Darwin Kernel Version 22.6.0: Fri Sep 15 13:41:28 PDT 2023; root:xnu-8796.141.3.700.8~1/RELEASE_ARM64_T6020 arm64
```

aykevl, 2023-09-26:
I suspect you're missing Rosetta 2. Installing that should get things working.

Right now the Homebrew version is still amd64 only (because of missing darwin/arm64 CI to build release artifacts).

deadprogram, 2024-05-12:
Now that there is an native darwin arm64 release, this problem is gone, so closing. Thank you!

closed 2024-05-12 by deadprogram