tinygo-org/tinygo #3942: Esp32 webcam

BrunoTeixeira1996, 2023-10-05:
Does tinygo support esp32 with a small câmera like [this](https://pt.aliexpress.com/item/1005005295618613.html?src=google&src=google&albch=shopping&acnt=708-803-3821&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=19373920468&albag=&trgt=&crea=pt1005005295618613&netw=x&device=m&albpg=&albpd=pt1005005295618613&gclid=CjwKCAjwvfmoBhAwEiwAG2tqzD0HRsyTrJ8-b8X4LAYsZhtbSI52frcVEMXDJIiQu11LIovGkuLc-xoCiTAQAvD_BwE&gclsrc=aw.ds&aff_fcid=b05dc5acd18c4443a08785ba6986f603-1696516902081-08256-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=b05dc5acd18c4443a08785ba6986f603-1696516902081-08256-UneMJZVf&terminal_id=38d8b8f2571c432d84e9f5dd900d5d90&afSmartRedirect=y)?

I tried to find in the docs but i was not able to find it.

Gustavomurta, 2023-10-17:
I also didn't find this ESP32cam project on TinyGo. But if it works with Arduino or C++, it is also possible in TinyGo.

_https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer_

BrunoTeixeira1996, 2023-10-17:
@Gustavomurta I would love to implement this in TinyGo but I have no idea on how to start to be fair. That link contains some header files that should be present in TinyGo and I don't think thats available

Gustavomurta, 2023-10-17:
I'm currently studying and learning Tinygo too. Rs rs. 
I hope to reach the level of knowledge to develop packages.

BrunoTeixeira1996, 2023-10-17:
@Gustavomurta Ill wait then hehe!