tinygo-org/tinygo #3943: cobra tests fail with "unimplemented: (reflect.Type).NumOut()"

dkegel-fastly, 2023-10-06:
Redirecting from https://github.com/tinygo-org/tinygo/issues/2660#issuecomment-1749948657

Cobra's relatively popular, but doesn't pass tests in tinygo 0.28.1 or 0.29.0:

```
$ tinygo version
tinygo version 0.28.1 darwin/amd64 (using go version go1.20.7 and LLVM version 15.0.0)
$ git clone git@github.com:spf13/cobra.git
$ cd cobra
$ tinygo test ./...
go: downloading github.com/cpuguy83/go-md2man/v2 v2.0.2
go: downloading github.com/russross/blackfriday/v2 v2.1.0
panic: unimplemented: (reflect.Type).NumOut()
FAIL	github.com/spf13/cobra	0.330s
ok  	github.com/spf13/cobra/doc	0.304s
```

dgryski, 2023-10-11:
Adding support for reflection information for functions is unlikely to appear anytime soon.

davecheney, 2026-10-27:
Thanks for reporting this and providing the reproduction steps. The missing function-signature reflection support, including `reflect.Type.NumOut`, is already tracked in #2650.

Closing this as a duplicate so we can keep progress in one place.

closed 2026-10-27 by davecheney