tinygo-org/tinygo #3944: Build fails with: "ld.lld: error: undefined symbol"

bitwombat, 2023-10-08:
Command: `go build main.go`
OS: Ubuntu 20.04
Go version: 1.20.2
Arch: x64
Not sure what else would be helpful, please let me know.
[See attached log](https://github.com/tinygo-org/tinygo/files/12839927/log.txt)

aykevl, 2023-10-08:
Looks like you are importing lots of packages that use Go assembly. This is not currently supported (see #3103).

You may be able to work around it using `-tags=purego`.

bitwombat, 2023-10-09:
Thanks for the reply @aykevl . Sorry, I had looked for similar issues and somehow didn't find #3103. `-tags=purego` did not make a difference.

It's a tiny module. I'm only using stdlib except for my test.

```
module github.com/bitwombat/tag

go 1.20

require github.com/stretchr/testify v1.8.4

require (
	github.com/davecgh/go-spew v1.1.1 // indirect
	github.com/pmezard/go-difflib v1.0.0 // indirect
	gopkg.in/yaml.v3 v3.0.1 // indirect
)
```

aykevl, 2023-10-10:
Here is an example that is not yet:

```
tinygo:ld.lld: error: undefined symbol: crypto/subtle.xorBytes
>>> referenced by xor.go:22 (/home/gbell2/.asdf/installs/golang/1.20.2/go/src/crypto/subtle/xor.go:22)
>>>               /home/gbell2/.cache/tinygo/thinlto/llvmcache-175C3A127B427402CAD826DA1277D681B2B45CD6:(crypto/subtle.XORBytes)
>>> referenced by xor.go:22 (/home/gbell2/.asdf/installs/golang/1.20.2/go/src/crypto/subtle/xor.go:22)
>>>               /home/gbell2/.cache/tinygo/thinlto/llvmcache-175C3A127B427402CAD826DA1277D681B2B45CD6:((*crypto/cipher.gcm).auth)
```

So it looks like you're importing some crypto packages (that use assembly for speed), but this isn't supported yet in TinyGo. (I don't know whether #3103 will fix it, but it should reduce the number of link errors at least).

dgryski, 2023-10-10:
In 1.21, the xorbytes was replaced with a a package with a pure-Go implementation selectable with `-tags=purego`.

bitwombat, 2023-10-11:
Using go 1.21.1 and `-tags=purego`, same result.

>So it looks like you're importing some crypto packages 

Yeah, weird, not directly. 

It's one of these:

```
"bytes"
"encoding/json"
"fmt"
"io"
"log"
"net/http"
"os"
"strings"
"text/template"
``` 

I'll make a minimal testcase, if that's useful.

dgryski, 2023-10-11:
`net/http` imports the crypto packages that require the xorbytes code.

bitwombat, 2023-10-15:
Thanks for the explanations.

closed 2023-10-15 by bitwombat