tinygo-org/tinygo #3956: type C.bool  is undfined

Patrick0308, 2023-10-16:
# Environment:
tinygo version: 0.30.0
operating System: macOS
architecture: aarch64

# Description:
 Compilation brakes with an undefined C._Bool

# Steps to reproduce:
Given the following Go-code:
```
package main

/*
  #include "stdbool.h"
*/
import "C"

import "fmt"

func main() {
	fmt.Printf("%s", C.bool(true))
}

```
Run

`tinygo run main.go`

# Expected behavior:
```
❯ tinygo run main.go
%!s(main._Ctype__Bool=true)%    
```

# Actual behavior:

```
❯ tinygo run main.go
# command-line-arguments
../../../../opt/homebrew/Cellar/tinygo/0.30.0/lib/clang/include/stdbool.h:20:14: undefined: C._Bool
```

Gustavomurta, 2023-10-17:
I´m not SW expert, but searching this argument - #include "stdbool.h" , I found this:

#include <stdio.h>  (this syntax is different)

Patrick0308, 2023-10-20:
https://gcc.gnu.org/onlinedocs/cpp/Search-Path.html

The difference between "stdbool.h" and <stdbool.H> is searching path. However it doesn't matter on this case.

deadprogram, 2023-10-23:
This was corrected in https://github.com/tinygo-org/tinygo/pull/3928 which will be part of the next release.

Patrick0308, 2023-10-24:
@deadprogram I tested with dev branch.  This case failed yet.

deadprogram, 2024-02-28:
Completed with `v0.31.0` so now closing. Thank you!

closed 2024-02-28 by deadprogram