tinygo-org/tinygo #3965: Wasm inside vscode extension

emil14, 2023-10-19:
Turns out this line

```ts
import "../wasm/wasm_exec.js";
```

Leads to such errors in debugger console if you gonna debug your vscode extension

```
hello from resolveCustomTextEditor
"go-to-character-position" is active
TypeError: Cannot read properties of undefined (reading '0')
	at f (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:158753)
	at v (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:159041)
	at 4441 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:137659)
	at __webpack_require__ (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:1621368)
	at 8039 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:159308)
	at __webpack_require__ (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:1621368)
	at 8160 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:154661)
	at __webpack_require__ (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:1621368)
	at 8114 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:108456)
	at __webpack_require__ (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:1621368)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:290197
	at async /Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:290161
	at async t.oneDataSystemClientFactory (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/git/dist/main.js:2:290132) {stack: 'TypeError: Cannot read properties of undefine…ces/app/extensions/git/dist/main.js:2:290132)', message: 'Cannot read properties of undefined (reading '0')'}
TypeError: Cannot read properties of undefined (reading '0')
	at h (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:91271)
	at b (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:91559)
	at 4441 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:70177)
	at a (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:1129043)
	at 8039 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:91826)
	at a (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:1129043)
	at 8160 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:87179)
	at a (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:1129043)
	at 779 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:41040)
	at a (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:1129043)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:317919
	at async /Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:317883
	at async t.oneDataSystemClientFactory (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github/dist/extension.js:2:317854) {stack: 'TypeError: Cannot read properties of undefine…extensions/github/dist/extension.js:2:317854)', message: 'Cannot read properties of undefined (reading '0')'}
Checking whether to register object drop extension
Registering object drop provider
Registered object drop provider
TypeError: Cannot read properties of undefined (reading '0')
	at m (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:93792)
	at y (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:94080)
	at 4441 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:72698)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:728117)
	at 8039 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:94347)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:728117)
	at 8160 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:89700)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:728117)
	at 8114 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:43495)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:728117)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:225236
	at async /Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:225200
	at async t.oneDataSystemClientFactory (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/merge-conflict/dist/mergeConflictMain.js:2:225171) {stack: 'TypeError: Cannot read properties of undefine…-conflict/dist/mergeConflictMain.js:2:225171)', message: 'Cannot read properties of undefined (reading '0')'}
TypeError: Cannot read properties of undefined (reading '0')
	at m (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:91395)
	at y (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:91683)
	at 54441 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:70258)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:1088597)
	at 8039 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:91951)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:1088597)
	at 98160 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:87298)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:1088597)
	at 50779 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:41061)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:1088597)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:223072
	at async /Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:223036
	at async t.oneDataSystemClientFactory (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/typescript-language-features/dist/extension.js:2:223007) {stack: 'TypeError: Cannot read properties of undefine…language-features/dist/extension.js:2:223007)', message: 'Cannot read properties of undefined (reading '0')'}
[... 8 lines cut ...]
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/json-language-features/client/dist/node/jsonClientMain.js:2:1095355)
	at 1616 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/json-language-features/client/dist/node/jsonClientMain.js:2:89670)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/json-language-features/client/dist/node/jsonClientMain.js:2:1095355)
	at 8604 (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/json-language-features/client/dist/node/jsonClientMain.js:2:43493)
[... cut ...]
	at async /Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github-authentication/dist/extension.js:2:230610
	at async t.oneDataSystemClientFactory (/Applications/Visual Studio Code.app/Contents/Resources/app/extensions/github-authentication/dist/extension.js:2:230581) {stack: 'TypeError: Cannot read properties of undefine…ub-authentication/dist/extension.js:2:230581)', message: 'Cannot read properties of undefined (reading '0')'}
```

I'm trying to embed wasm file compiled with tinygo into my vscode extension and I don't need WASI

aykevl, 2024-03-16:
You are importing TinyGo compiled wasm files inside your VSCode extension?
This was never tested. If there are bugs that need fixing in wasm_exec.js feel free to make a PR.

davecheney, 2026-10-26:
Sorry for the delay in replying. Is this still a problem with TinyGo 0.43 or later? If so, please share a minimal extension that reproduces the failure, your TinyGo and VS Code versions, and the command used to build the WebAssembly module.