tinygo-org/tinygo #3966: "-tags" option should support a comma separated list.

nobonobo, 2023-10-20:
In the Go standard, parameter format for the "-tags" option are comma-separated.

> -tags tag,list
	a comma-separated list of additional build tags to consider satisfied
	during the build. For more information about build tags, see
	'go help buildconstraint'. (Earlier versions of Go used a
	space-separated list, and that form is deprecated but still recognized.)

Exar04, 2023-11-24:
if i am not wrong making following changes in loader/list.go line number 22 will solve the issue, right?
```
args = append(args, "-tags", strings.Join(config.BuildTags(), " "))
```
changing it to
```
args = append(args, "-tags", strings.Join(config.BuildTags(), ","))
```

aykevl, 2023-11-25:
@Exar04 no that's just for how to call `go list` (although it might be a good idea to make that change too). We need to change how build tags are parsed by TinyGo.

omnide, 2024-03-05:
I updated linked PR #4088 with the required golang/tools dependency updates needed to resolve this.  My upstream fix to golang/tools was merged before the v0.18.0 tools release cut in mid February.

buildutil fix upstream is:  https://github.com/golang/tools/commit/5f9069195d793b7284459955e7e1b0283abab71c