tinygo-org/tinygo #3970: wasmtime 14 changes --mapdir to --dir

ydnar, 2023-10-21:
…breaking tests with `-target=wasi`: See https://github.com/bytecodealliance/wasmtime/pull/7301

> This commit changes the --dir argument on the wasmtime CLI to be HOST::GUEST rather than GUEST::HOST. This matches Docker for example and is a little more consistent with only --dir path where the first argument is always treated as a host directory.
>
> In terms of breaking-ness the movement from --mapdir to --dir hasn't been released with Wasmtime 14 yet so my hope is that this can land on both main and Wasmtime 14.0.0 before it's released to avoid any breakage other than existing scripts migrating from --mapdir to --dir.

Wasmtime 14 was released today, which broke `tinygo test -target=wasi`: https://github.com/bytecodealliance/wasmtime/blob/main/RELEASES.md#1400

Existing `--mapdir` arguments should be changed to:

```sh
      --dir <HOST_DIR[::GUEST_DIR]>
          Grant access of a host directory to a guest.
```

deadprogram, 2023-11-04:
Tagged to close this issue after next release.

dgryski, 2023-11-04:
We still need to update argument construction if we haven't already.

ydnar, 2023-11-04:
I think argument construction is correct now.

dgryski, 2023-11-04:
Ah yes, thanks.  Was trying to figure out why I was getting this warning from wasmtime.  Needed to rebase my branch against the latest `dev`:
```
~/go/src/github.com/dgryski/bug/intf $ tinygo run -target=wasi main.go
warning: this CLI invocation of Wasmtime will be parsed differently in future
         Wasmtime versions -- see this online issue for more information:
         https://github.com/bytecodealliance/wasmtime/issues/7384

         Wasmtime will now execute with the old (<= Wasmtime 13) CLI parsing,
         however this behavior can also be temporarily configured with an
         environment variable:

         - WASMTIME_NEW_CLI=0 to indicate old semantics are desired and silence this warning, or
         - WASMTIME_NEW_CLI=1 to indicate new semantics are desired and use the latest behavior
```

deadprogram, 2024-02-28:
Completed with `v0.31.0` so now closing. Thank you!

PR #3972 (2023-11-02): compileopts, targets, main: support Wasmtime v14
This PR adds support for [Wasmtime v14](https://github.com/bytecodealliance/wasmtime/blob/main/RELEASES.md#1400) (released 2023-10-20), reenabling support for:

`tinygo test -v -target=wasi ./...`

1. Changes `--mapdir=GUEST::HOST` to `--dir=HOST::GUEST`
2. Reorders CLI arguments to `wasmtime` so Wasmtime arguments are before the `.wasm` module argument, and arguments to the Go program (e.g. `-test.v`) are after. The `--` argument has no effect.

Note this PR does not attempt to detect the version of Wasmtime being run.

For additional context,  [... cut]
closed 2024-02-28 by deadprogram