tinygo-org/tinygo #3973: Validating signature of JWT gives `undefined symbol: crypto/internal/bigmod.addMulVVW1024`

rica-graca, 2023-10-23:
## Description

Currently, I am trying to validate the signature of a JWT inside a wasm filter that is compiled with tinygo. My code is the following:

```
import (
"github.com/golang-jwt/jwt/v4"
)

```
	parser := jwt.NewParser()
	claims := claims.AuthClaims{}
	signedToken, err := parser.Parse(authorizationToken, func(t *jwt.Token) (interface{}, error) {
		return tt, nil
	})
```

And the result of it is:

`tinygo:wasm-ld: error: /tmp/tinygo3147844765/main.o: undefined symbol: crypto/internal/bigmod.addMulVVW1536`

## Question
Is there any solution to this. Is there a way to validate a JWT signature without the need of this crypto library? I would be very grateful if someone has a solution. There should definitely be a way to verify a signature as in WASM code compiled with tinygo

lann, 2023-10-23:
This may solve your problem: https://github.com/tinygo-org/tinygo/issues/3919#issuecomment-1730312952

rica-graca, 2023-10-24:
Thank you so much @lann

closed 2023-10-24 by rica-graca