tinygo-org/tinygo #3980: undefined symbol crypto/internal/bigmod.addMul

zetaab, 2023-10-28:
code:

```
package main

import (
	"context"
	"fmt"

	"golang.org/x/oauth2"
)

func main() {
	conf := &oauth2.Config{
		ClientID:     "test",
		ClientSecret: "test",
	}

	token, err := conf.Exchange(context.Background(), "codehere")
	if err != nil {
		panic(err)
	}
	fmt.Println(token)
}
```

```
% tinygo build -o main -scheduler=asyncify -target=wasi main.go 
tinygo:wasm-ld: error: /var/folders/k1/7lyy6srx6x59z9j7z0_rs1fw0000gp/T/tinygo2745426231/main.o: undefined symbol: crypto/internal/bigmod.addMulVVW1024
tinygo:wasm-ld: error: /var/folders/k1/7lyy6srx6x59z9j7z0_rs1fw0000gp/T/tinygo2745426231/main.o: undefined symbol: crypto/internal/bigmod.addMulVVW1024
tinygo:wasm-ld: error: /var/folders/k1/7lyy6srx6x59z9j7z0_rs1fw0000gp/T/tinygo2745426231/main.o: undefined symbol: crypto/internal/bigmod.addMulVVW1536
tinygo:wasm-ld: error: /var/folders/k1/7lyy6srx6x59z9j7z0_rs1fw0000gp/T/tinygo2745426231/main.o: undefined symbol: crypto/internal/bigmod.addMulVVW1536
tinygo:wasm-ld: error: /var/folders/k1/7lyy6srx6x59z9j7z0_rs1fw0000gp/T/tinygo2745426231/main.o: undefined symbol: crypto/internal/bigmod.addMulVVW2048
tinygo:wasm-ld: error: /var/folders/k1/7lyy6srx6x59z9j7z0_rs1fw0000gp/T/tinygo2745426231/main.o: undefined symbol: crypto/internal/bigmod.addMulVVW2048
failed to run tool: wasm-ld
error: failed to link /var/folders/k1/7lyy6srx6x59z9j7z0_rs1fw0000gp/T/tinygo2745426231/main: exit status 1
```

How I could use Golang native oauth2 library with wasi? `go build -o main main.go` works fine. Is there any alternatives for oauth2?

dgryski, 2023-11-01:
The crypto code currently needs `-tags=purego` to build successfully with tinygo.

zetaab, 2023-11-01:
ooh! got it working! thanks a lot

closed 2023-11-01 by zetaab