tinygo-org/tinygo #3987: mingw expected in incorrect location (tinygo-macos)

coolaj86, 2023-11-04:
## Error

```sh
GOOS=windows GOARCH=amd64 tinygo build .
```

```text
error: open /Users/aj/.local/opt/tinygo-0.30.0/lib/mingw-w64/mingw-w64-headers/crt/_mingw.h.in: no such file or directory
```

## Actual Location

The actual location appears to be...

```sh
pushd ~/.local/opt/tinygo/
fd | rg 'mingw'
```

```text
~/.local/opt/tinygo/lib/mingw-w64/mingw-w64-headers/_mingw.h.in
```

coolaj86, 2023-11-04:
## Workaround

Go to the `mingw-w64-headers` and link itself to `crt`:

```sh
pushd ~/.local/opt/tinygo-0.30.0/lib/mingw-w64/mingw-w64-headers/
ln -s ./. ./crt
```

brettinternet, 2025-03-27:
I have the same issue but I've installed tinygo via Homebrew so my root paths are a little different:

```sh
GOOS=windows tinygo build .
open /opt/homebrew/Cellar/tinygo/0.37.0/lib/mingw-w64/mingw-w64-headers/crt/_mingw.h.in: no such file or directory
```

The actual path appears to be:

```
/opt/homebrew/Cellar/tinygo/0.37.0/lib/mingw-w64/mingw-w64-headers/_mingw.h.in
```