tinygo-org/tinygo #3988: Playdate support

pattmax00, 2023-11-04:
It would be nice to have [Playdate](https://play.date/) support. It is using a Cortex M7 (ARM) processor which is already supported:
https://github.com/tinygo-org/tinygo/issues/512
https://github.com/cmsis-svd/cmsis-svd/pull/82

A good example of a final C project that could maybe help as a reference would be this: https://github.com/ericlewis/playdate-tamagotchi since the final `.pdx` is quite simple. The `.pdx` is basically just a folder with
1. The compiled code in pdex.bin
2. A pdxinfo (just some text)
3. And then the the assets are built into `.pdi` files

You can extract the latest release yourself to see.

Even if we could just compile the `pdex.bin` would be nice.

[Here is the C API reference](https://sdk.play.date/2.1.1/Inside%20Playdate%20with%20C.html)

I think the GBA implementation is already performing some extra steps to build a `.gba` ROM file. So being able to compile for the Playdate and build a `.pdx` shouldn't be outside the scope of this project, but if it is feel free to let me know and close the issue, but I think the Playdate would be a very good candidate for TinyGo.

Playboy26, 2024-05-21:
[image]
Hey playdate support I just got my playdate handheld on May the 16th and then on May the 17th I was say up my playdate handheld and it was not looking too good so how can I fix my playdate handheld

pattmax00, 2024-05-21:
> [image]
> Hey playdate support I just got my playdate handheld on May the 16th and then on May the 17th I was say up my playdate handheld and it was not looking too good so how can I fix my playdate handheld 

We're not playdate support, and I'm not sure how this is relevant to the Tinygo feature request? This is a compiler for embedded Go development.

aykevl, 2024-09-23:
You would need to add support for the STM32F746 as well as specific board support for the Playdate.
A PR you could look at for how to do this, is this one: https://github.com/tinygo-org/tinygo/pull/4219