tinygo-org/tinygo #3990: UART.Configure() has incorrect signature in machine.

bgould, 2023-11-07:
https://github.com/tinygo-org/tinygo/blob/2b215955cac75b1308d02269b8ea5b73448708a9/src/machine/machine_nrf.go#L174

bgould, 2023-11-07:
Upon further investigation ... it appears `machine.Serialer` is protected by build tags in `src/machine/usb.go` ... so maybe isn't intended for UART to implement it?  Unless there is some detail I'm missing, I think that UART should implement `machine.Serialer`.

sago35, 2023-11-08:
The machine.Serialer interface, if I recall correctly, was introduced for the purpose of tidying up implementations such as -serial=uart. Therefore, it probably does not take into account environments without USB support.

As you are writing, I believe UART should implement machine.Serialer.

closed 2026-11-07 by deadprogram