tinygo-org/tinygo #3992: IsExported undefined (type reflect.Method has no field or method IsExported)

danielvladco, 2023-11-09:
Hello, thanks for your amazing work on tinygo!

I am facing an issue accessing `IsExported` method from the reflect package

## Minimal steps to reproduce:
1. Write this program:
```
package main

import (
	"reflect"
)

func main() {
	if reflect.TypeOf(0).Method(0).IsExported() {
		println("IsExported exists")
	}
}
```
2. Run `tinygo build .`

## Expected result
To compile and build

## Actual result
Error:
```
main.go:8:33: reflect.TypeOf(0).Method(0).IsExported undefined (type reflect.Method has no field or method IsExported)
```

fniephaus, 2025-06-18:
I ran into this problem as well when trying to compile https://github.com/xuri/excelize-wasm with TinyGo:

```
# github.com/tiendc/go-deepcopy
../../../go/pkg/mod/github.com/tiendc/go-deepcopy@v1.5.0/struct_copier.go:256:28: dM.IsExported undefined (type *reflect.Method has no field or method IsExported)
```

Working around this particular `IsExported()` call, I can successfully compile excelize-wasm to Wasm.

Would be great if this could be fixed!

aykevl, 2025-09-30:
This should be fixed now, please comment if it isn't.