tinygo-org/tinygo #3994: "Uncaught (in promise) TypeError: wasm function signature contains illegal type" in chrome 84

kingchn, 2023-11-11:
use  tinygo0.30.0

error in **chrome 84**:
Uncaught (in promise) TypeError: wasm function signature contains illegal type

```
package main

import "syscall/js"

//export add
func add(x, y uint32) uint32 {
	//jsCode := `aaa`
	//js.Global().Get("alert").Invoke(jsCode)
	return x + y
}

func add2(this js.Value, inputs []js.Value) interface{} {
	a := inputs[0].Int()
	b := inputs[1].Int()
	return a + b
	//return "55"
}

func main() {
	wait := make(chan struct{}, 0)
	js.Global().Set("add2", js.FuncOf(add2))
	<-wait
}

```

use tinygo0.30.0.windows-amd64
tinygo build -o test_wasm_with_js.wasm -target **wasm** ./main.go

deadprogram, 2024-12-02:
@kingchn is this still an issue with the latest versions of TinyGo?

ZhenmingMa, 2025-02-25:
I hava the same question.

davecheney, 2026-10-27:
TinyGo 0.44-dev (`b7ec945f`) builds and runs this example on Node 26. It exercises both a Go function exported to JavaScript and a Go callback registered with `js.FuncOf`.

`main.go`:

```go
package main

import "syscall/js"

func main() {}

//export registerCallback
func registerCallback() {
	js.Global().Set("add2", js.FuncOf(func(this js.Value, inputs []js.Value) any {
		return inputs[0].Int() + inputs[1].Int()
	}))
}

//export add
func add(x, y uint32) uint32 {
	return x + y
}
```

Build with:

```sh
tinygo build -target=wasm -buildmode=c-shared -o main.wasm main.go
```

Using TinyGo’s matching `wasm_exec.js`, this Node host instantiates the module and calls both functions:

```js
const fs = require("node:fs");
const { webcrypto } = require("node:crypto");

globalThis.crypto ??= webcrypto;
globalThis.require = require;
globalThis.fs = fs;
globalThis.performance = { now: () => Number(process.hrtime.bigint()) / 1e6 };

require(`${process.env.TINYGOROOT}/targets/wasm_exec.js`);

async function main() {
	const go = new Go();
	const { instance } = await WebAssembly.instantiate(
		fs.readFileSync("main.wasm"),
		go.importObject,
	);
	await go.run(instance);

	console.log("exported add:", instance.exports.add(20, 22));
	instance.exports.registerCallback();
	console.log("registered JS callback:", globalThis.add2(20, 22));
}

main().catch((err) => {
	console.error(err);
	process.exitCode = 1;
});
```

Output:

```text
exported add: 42
registered JS callback: 42
```

Chrome 84 is no longer available to verify the original browser-specific error, so this test does not establish whether that exact failure is fixed. Does the issue still occur with a current TinyGo release in a browser you can test now? If so, please share the browser version and a minimal reproducer.