tinygo-org/tinygo #3997: What kind of GC is `-gc=precise`?

vlkv, 2023-11-16:
Hello!
There is no info about `-gc=precise` option on https://tinygo.org/docs/reference/usage/important-options/ page. But such option definitely exists, because of https://github.com/tinygo-org/tinygo/blob/731532cd2b6353b60b443343b51296ec0fafae09/compileopts/options.go#L11 

tinygo --help also says not much:
```
  -gc string
    	garbage collector to use (none, leaking, conservative)
```    

Can you please explain what is this kind of GC and what are its pros/cons (compared to 'conservative')? 

Thank you.

vlkv, 2024-05-31:
Please, can somebody respond?..

aykevl, 2024-06-07:
It's almost the same as the conservative GC, but tracks which area of memory definitely aren't pointers. So it has a smaller chance of false positives (which means reduced memory consumption) and may or may not be faster (depending on your use case). It's still mostly conservative, but a bit smarter.

This should really be added to the website.