tinygo-org/tinygo #3998: TinyGo for ESP32 on Windows 10 - not compiling

Gustavomurta, 2023-11-17:
I can't compile TinyGo for ESP32 on Windows 10. I saw that to run the esptool tool on Linux, **esptool.py** is used. But in Windows 10 there is no such command, after installing esptool. The command is **esptool** (without .py).

`C:\Users\jgust>esptool version`
`esptool.py v4.6.2`

Therefore, when trying to compile TinyGo for ESP32 I receive this error message.

**"esptool.py": executable file not found in %PATH%**

```
PS C:\Users\jgust\tinygo\programas\esp32\flash_spi\flash_example\console\spi> tinygo flash -target=esp32-coreboard-v2 main.go
error: failed to flash C:\Users\jgust\AppData\Local\Temp\tinygo4028449839\main.bin: exec: "esptool.py": executable file not found in %PATH%
```

```
C:\Users\jgust>tinygo env
GOOS="windows"
GOARCH="amd64"
GOROOT="C:\\Program Files\\Go\\"
GOPATH="C:\\Users\\jgust\\go\\"
GOCACHE="C:\\Users\\jgust\\AppData\\Local\\tinygo"
CGO_ENABLED="1"
TINYGOROOT="C:\\Users\\jgust\\scoop\\apps\\tinygo\\current"
```

Can anyone help with this problem? Thank you.

sago35, 2023-11-17:
Copying `esptool.exe` to create `esptool.py.exe` should work fine.

The following is reference information, but it is written in Japanese.
https://qiita.com/sago35/items/4517cc96919f84f0efb3#esptoolpy-%E3%81%AE%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB-%E3%83%90%E3%82%A4%E3%83%8A%E3%83%AA%E3%82%92%E5%85%A5%E6%89%8B%E3%81%99%E3%82%8B

sago35, 2023-11-17:
I think it's better to modify the main TinyGo codebase.

Gustavomurta, 2023-11-17:
Sago35, I copied the **esptool** program from the link you indicated:
_https://github.com/espressif/esptool/releases/_

To be able to use it on Windows, I renamed the program to **esptool.py.exe**

I found where the original tool was in Windows:
`C:\Users\jgust\AppData\Local\Programs\Python\Python312\Scripts`

I copied the renamed tool into this folder. I tested it and it worked! Thank you very much.

It's not the ideal solution, but it's an alternative.

davecheney, 2026-10-26:
Thanks for the report. ESP32 flashing is now built into TinyGo via https://github.com/tinygo-org/tinygo/pull/5229, so installing or renaming `esptool.py` is no longer necessary.

```sh
tinygo flash -target=esp32-coreboard-v2 -monitor examples/serial
```

closed 2026-10-26 by davecheney