tinygo-org/tinygo #4002: lld: `inline asm clobber` warnings when building a program for the RP2040

soypat, 2023-11-22:
Follow up on discussion over at [slack](https://gophers.slack.com/archives/CN4R2DQV7/p1700508003394059). I've managed to narrow down what seems to be a compiler bug to the attached MWE. A few observations:

- When the program does compile after modifying it slightly, a previously "clobbered" version will compile succesfully. `tinygo clean` undoes this benefit
- When compiling program succesfully for different architecture, i.e linux/amd64, a previously "clobbered" program will succesfully compiler. `tinygo clean` undoes this 

```shell
# tinygo version 0.31.0-dev-d4189fec linux/amd64 (using go version go1.21.4 and LLVM version 16.0.1)
tinygo build -target=pico ./examples/llvm-bug/
```

```go
package main

import (
	"github.com/soypat/seqs/stacks" // v0.0.0-20231122002606-b18ce4c4b639
)

func main() {
	defer func() {
		println("main: exiting")
	}()
	stack := stacks.NewPortStack(stacks.PortStackConfig{})
	go NICLoop()
	var dhcp stacks.DHCPv4Client
	for {
		stack.OpenUDP(68, dhcp.HandleUDP)
		stack.FlagPendingUDP(68) // Force a DHCP discovery.
	}
}

func NICLoop() {
	var queue [4*2030]byte
	print(queue[:])
}

```

deadprogram, 2024-12-13:
@soypat is this still in issue in the latest `dev` branch?

soypat, 2024-12-15:
I was not able to reproduce this on dev branch. I guess we can close :)

closed 2024-12-15 by deadprogram