tinygo-org/tinygo #4004: Cannot read properties of undefined (reading 'exports')

jzyzxx, 2023-11-24:
### go code:
`
package main

import (
	"crypto/rand"
	"crypto/rsa"
	"crypto/x509"
	"encoding/pem"
)

var priKey = []byte(`-----BEGIN RSA PRIVATE KEY-----
MIICXQIBAAKBgQCe8zGb4UAMg2A63pH+/W145hHvYQPJlkX6OfzJ1215htCI6Pyh
2TdHRrDqVU6wP609ao9tLxRsbbXrajBGXiq2ijRX7AKrsVdhYi2J+B2q/CrsH5CD
Ka16YCVPPwf/oZDz/hxrcjZjhOoSIZupY3/xzOBTTjcVcvWbTxGw0wOm6wIDAQAB
AoGABrVg7KFPILgSwalPJCHyEt4y95VyoXl0LqFv59ztw+lKt9yNfQ875Ag5w0oi
bhHh7+ulbghEpmbi/LKYov+qccTQMCz4PW1g85LrUYI5PaGKQfsTAWldQeV/mxCk
mimCk8bahoWPX4i2fnyFdCCn7f3kL8RqRp4NXu2En2gJkPECQQDL3QZrRBpxuE8L
vgMPNew+II3XtiMzsXc/EwHpAT2hY/pOXt0pvtGfAU2d1JSzmHlBfqPkhr2S0obE
PpdsXyG3AkEAx5mt8rsDflY8vRYU7Xao0+Smt+9ujMhvtzzS9W62VCUU8xc0UG+x
umgxofSOedkoaR7k2jqFYYbC1CrwPyAUbQJBALle2R9gZctSFE5REOcb2R0E7PVg
oNG4ZP3tgqckga3nAwuQJvp2kJVM0g7Z5f0If/mV9eEuw+JlnDWF1JquRjECQQCi
ZrT0eRsnkO0MgEn4yAInnbPUlphhLbhP48pVbYYmQqGgBHJJPAfkfmBbwMqn83uA
xGU59kGOD4K39FPTWLulAkAngU3Yv8vYmZKcYXuc/TZjxa0sMuRVroWO6ciW81so
+sFpf0SM9Ysgf/nKtux7juJABCfF1ffDQdKwederSMOc
-----END RSA PRIVATE KEY-----`)

var pubKey = []byte(`-----BEGIN PUBLIC KEY-----
MIGfMA0GCSqGSIb3DQEBAQUAA4GNADCBiQKBgQCe8zGb4UAMg2A63pH+/W145hHv
YQPJlkX6OfzJ1215htCI6Pyh2TdHRrDqVU6wP609ao9tLxRsbbXrajBGXiq2ijRX
7AKrsVdhYi2J+B2q/CrsH5CDKa16YCVPPwf/oZDz/hxrcjZjhOoSIZupY3/xzOBT
TjcVcvWbTxGw0wOm6wIDAQAB
-----END PUBLIC KEY-----`)
func main()  {
   
}

//export RSAEncrypt
func RSAEncrypt(origData string) string {
	block, _ := pem.Decode(pubKey)
	pubInterface, _ := x509.ParsePKIXPublicKey(block.Bytes)
	pub := pubInterface.(*rsa.PublicKey)
	bits, _ := rsa.EncryptPKCS1v15(rand.Reader, pub, []byte(origData))
	return string(bits)
}

//export RSADecrypt
func RSADecrypt(cipherTxt string) string {
	block, _ := pem.Decode(priKey)
	pri, _ := x509.ParsePKCS1PrivateKey(block.Bytes)
	bits, err := rsa.DecryptPKCS1v15(rand.Reader, pri, []byte(cipherTxt))
    if err != nil {
        panic(err)
    }
	return string(bits)
}
`
### build params：
`
tinygo build -o ./wasm.wasm -no-debug -panic=trap -scheduler=none -target wasm .\main.go
`
### Some errors accured in browser:
`wasm_exec_tiny.js:223 Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'exports')
    at loadSlice (wasm_exec_tiny.js:223:38)
    at random_get (wasm_exec_tiny.js:283:30)
    at arc4random_buf (wasm.wasm:0x58c)
    at io.ReadFull (wasm.wasm:0x5313)
    at RSAEncrypt.command_export (wasm.wasm:0x3c71b)
    at test.html:20:41
`

jzyzxx, 2023-11-24:
My tinygo version is 0.30.0

CrazyXi, 2024-04-10:
maybe you should remove it -scheduler=none

rellfy, 2024-11-05:
Hi @jzyzxx, were you able to get this to work?
I raised an issue I had when working with the same libs: https://github.com/tinygo-org/tinygo/issues/4583