tinygo-org/tinygo #4029: Board Support: ESP32-C6

johanvandeweerd, 2023-12-06:
Hi

Just started to get into the ESP32 rabbit hole and bought myself a ESP32-C6 development board. Given that my experience in C/C++ is limited and I have quite some experience in Go, I would prefer to program the board using Tinygo.

After installing Tinygo, creating a simple HelloWorld main and flashing it to the board using Tinygo, I get following output:

```
# tinygo flash --target=esp32 --port=/dev/cu.usbmodem21401 main.go      
esptool.py v4.6.2
Serial port /dev/cu.usbmodem21401
Connecting...

A fatal error occurred: This chip is ESP32-C6 not ESP32. Wrong --chip argument?
error: failed to flash /var/folders/0s/snyz0vdx62s099y165bsy1wh0000gn/T/tinygo4172958918/main.bin: exit status 2
```

This is my `main.go` file

```
package main

import "fmt"

func main() {
	fmt.Println("Hello world!")
}
```

I assume it means that the ESP32-C6 boards are not yet supported? I looked at the [Adding a new board](https://github.com/tinygo-org/tinygo/wiki/Adding-a-new-board) page but got very lost very fast.

Do I need to wait for Tinygo to support this board (or can I help in any shape or form)?

deadprogram, 2023-12-07:
The first step would be getting an SVD file for that processor into https://github.com/cmsis-svd/cmsis-svd-data

Perhaps open an issue on that repo to see if anyone is working on that part yet?

Once there is an SVD files we can start on integration for TinyGo.

aykevl, 2023-12-07:
Official Espressif SVD files can be found here: https://github.com/espressif/svd/tree/main/svd

I suspect the ESP32-C6 will be similar to the ESP32-C3. You can grep for `esp32c3` to see where the build tag is defined. It might be as easy as copy-pasting the esp32c3.json file and modifying it to use `esp32c6` instead, but there is a good chance some parts need to be modified.

pcting, 2024-04-16:
@aykevl looks like an svd file is available for esp32c6 now. i can take a stab at trying to build support. is there a tool that converts the `.svd` file to `.ld`?

gandarez, 2025-03-09:
Hi folks, any update on this? I'm able to help you with this development.

deadprogram, 2025-03-11:
The SVD files for the c6 are already in the https://github.com/cmsis-svd/cmsis-svd-data repo that TinyGo uses to generate the wrappers, so nothing to stop the rest of the work on this since the `src/device/esp/esp32c6.go` is already being generated now. 😄

davecheney, 2026-10-27:
ESP32-C6 support is now available in TinyGo. I verified this with the XIAO ESP32-C6 target using this minimal LED program:

```go
package main

import "machine"

func main() {
	machine.LED.Configure(machine.PinConfig{Mode: machine.PinOutput})
	machine.LED.Set(true)
}
```

Build it with `tinygo build -target=xiao-esp32c6 main.go`. This compiled successfully with the current TinyGo tree. Please feel free to reopen this issue if you run into any trouble.

closed 2026-10-27 by davecheney