tinygo-org/tinygo #4036: Documentation is terrible

JohnRichard, 2023-12-10:
This page: https://tinygo.org/docs/guides/build/#with-a-self-built-llvm

Is so convoluted and confusing it is unusable.  Whoever wrote this document needs to go back and either link to LLVM build/compile page on the LLVM website, or write explicit instruction on how to make this work.   Right now, I am chasing down how to compile LLVM from source on my own, which is fine.  Clear, concise steps to reproduce should be made in this document if tinygo maintainers wish people to use their product.  

I would love to test out tinygo on my own projects, but right now, I have to do many other steps just to get it to work, without documentation support.  Bad form.  Please correct.

denysvitali, 2023-12-10:
```
make llvm-source llvm-build
```

works for me - is there any particular error you're seeing?

JohnRichard, 2023-12-10:
Call me dumb. I have tried to understand what I am supposed to do following this documentation, but since there are no literal steps its difficult for me to understand. 

I have tried that 'make' command in every conceivable folder and it just keeps telling me that there are no make files to use.  But its ok, because I am now compiling llvm myself following other documentation. 

The real issue I have with the documentation is its lack of explicit instruction and lack of explanation.  Just because everyone else knows exactly what's going on, doesn't mean I do. and I don't, I'm sure there are more who also do not. 

Give me information overload.  Don't tell me to type random commands. Explain what I am doing. 
That's shouldn't be hard.

denysvitali, 2023-12-10:
> Just because everyone else knows exactly what's going on, doesn't mean I do. and I don't, I'm sure there are more who also do not.

Yeah, I'm not arguing on this part - the documentation of this project could definitely be better - and lots of things are complex / undocumented. I've been in your exact same situation a while ago.
I hope I can help you further - but the thing you're most likely missing is:
- fully cloning the project (make sure to clone the git submodules too)
- understanding why you need to compile LLVM (unless you're developing for Xtensa - that is the Espressif chips you don't need it)

If you really want to go down the LLVM way - make sure you are running the `make` command from above from the root directory of this project

JohnRichard, 2023-12-13:
I build LLVM 18 by hand, it took hours. 
Then I built TinyGo, with the help of Google Bard. 
I think it works now. Someone really needs to work on clear documentation for the dumb (like me).

dgryski, 2023-12-13:
Can you give us the instructions you got from Google Bard you feel are missing from the documentation?

bgould, 2023-12-16:
> Clear, concise steps to reproduce should be made in this document if tinygo maintainers wish people to use their product.

To clarify, TinyGo is a _project_ not a _product_.

Most of the people who are part of this project are not paid for it. Even those who are paid to work on TinyGo as part of their job are often working on features and helping people with issues beyond what their job or employer demands.

There is a big problem in the open-source community with maintainers becoming burnt out because of the sometimes entitled/aggressive/abusive language and behavior from users of free software. Please do not contribute to that.

> No, it doesn't. I had to make gen-devices first, or the llvm-sources failed. Which is fine. Make at least tells you what the problem is, most of the time. Additionally, making llvm-sources and llvm-build were _required_ by make when building tinygo. They were not "additional steps", they were not a thing that needed to be done before tinyGo works; they are required before tinyGo even compiles.

The document only says that llvm-sources is an additional step if you have not already run it (i.e., if using a system-installed LLVM).  If you feel this could be expressed more clearly, you can submit a PR to the `dev` branch in this repository: https://github.com/tinygo-org/tinygo-site

I just tested building TinyGo from source with a self-built LLVM on Debian by pasting the commands from the documentation into my terminal verbatim, and was able to build and run TinyGo without having run `make gen-device`.  

```
$ make
CGO_CPPFLAGS="-I/home/bcg/src/upstream/tinygo-src/llvm-project/llvm/include -I/home/bcg/src/upstream/tinygo-src/llvm-build/include -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/home/bcg/src/upstream/tinygo-src/llvm-build/tools/clang/include -I/home/bcg/src/upstream/tinygo-src/llvm-project/clang/include -I/home/bcg/src/upstream/tinygo-src/llvm-project/lld/include" CGO_CXXFLAGS="-std=c++17" CGO_LDFLAGS="-L/home/bcg/src/upstream/tinygo-src/llvm-build/lib -lclang -Wl,--start-group -lclangAnalysis -lclangAST -lclangASTMatchers -lclangBasic -lclangCodeGen -lclangCrossTU -lclangDriver -lclangDynamicASTMatchers -lclangEdit -lclangExtractAPI -lclangFormat -lclangFrontend -lclangFrontendTool -lclangHandleCXX -lclangHandleLLVM -lclangIndex -lclangLex -lclangParse -lclangRewrite -lclangRewriteFrontend -lclangSema -lclangSerialization -lclangSupport -lclangTooling -lclangToolingASTDiff -lclangToolingCore -lclangToolingInclusions -Wl,--end-group -lstdc++ -Wl,--start-group -llldCOFF -llldCommon -llldELF -llldMachO -llldMinGW -llldWasm -Wl,--end-group -L/home/bcg/src/upstream/tinygo-src/llvm-build/lib  -lLLVMWindowsManifest -lLLVMWindowsDriver -lLLVMMCJIT -lLLVMLTO -lLLVMPasses -lLLVMIRPrinter -lLLVMExtensions -lLLVMLibDriver -lLLVMOption -lLLVMInterpreter -lLLVMFrontendHLSL -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMOrcTargetProcess -lLLVMOrcShared -lLLVMCoverage -lLLVMCoroutines -lLLVMXtensaDisassembler -lLLVMXtensaAsmParser -lLLVMXtensaCodeGen -lLLVMXtensaDesc -lLLVMXtensaInfo -lLLVMAVRDisassembler -lLLVMAVRAsmParser -lLLVMAVRCodeGen -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyUtils -lLLVMWebAssemblyInfo -lLLVMRISCVTargetMCA -lLLVMRISCVDisassembler -lLLVMRISCVAsmParser -lLLVMRISCVCodeGen -lLLVMRISCVDesc -lLLVMRISCVInfo -lLLVMAArch64Disassembler -lLLVMAArch64AsmParser -lLLVMAArch64CodeGen -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMARMDisassembler -lLLVMARMAsmParser -lLLVMARMCodeGen -lLLVMipo -lLLVMVectorize -lLLVMLinker -lLLVMFrontendOpenMP -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMX86TargetMCA -lLLVMMCA -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMCFGuard -lLLVMGlobalISel -lLLVMX86Desc -lLLVMX86Info -lLLVMMCDisassembler -lLLVMSelectionDAG -lLLVMInstrumentation -lLLVMAsmPrinter -lLLVMCodeGen -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMObjCARCOpts -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMSymbolize -lLLVMDebugInfoPDB -lLLVMDebugInfoMSF -lLLVMDebugInfoDWARF -lLLVMObject -lLLVMTextAPI -lLLVMMCParser -lLLVMIRReader -lLLVMAsmParser -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMBitReader -lLLVMCore -lLLVMRemarks -lLLVMBitstreamReader -lLLVMBinaryFormat -lLLVMTargetParser -lLLVMSupport -lLLVMDemangle -lrt -ldl -lm -lstdc++ "  go build -buildmode exe -o build/tinygo -tags "byollvm osusergo" -ldflags="-X github.com/tinygo-org/tinygo/goenv.GitSha1=`git rev-parse --short HEAD`" .
$ ./build/tinygo version
tinygo version 0.31.0-dev-731bd5cd linux/amd64 (using go version go1.20.6 and LLVM version 16.0.1)
$ tinygo run examples/serial
hello world!
hello world!
hello world!
^C
```

I have also recently built TinyGo from source on Windows using that same document. 

You have not shared basic information about the problems you encountered such as details about your build environment or specific errors that you received, and I think it will be very difficult for anyone to help with the things you'd like to see addressed.

Finally ... building from source using self-built LLVM is literally the most complicated way to install and use TinyGo, and for the most part is only necessary for specific/advanced use cases.  For most users, the releases that are distributed as binary downloads, container images, or via package managers are the easiest and most reliable way to use TinyGo.  It is also recommended in the document to use system-installed LLVM when possible.  So I don't think the title of the issue "Documentation is terrible" is either accurate or appropriate.

JohnRichard, 2023-12-16:
@So I don't think the title of the issue "Documentation is terrible" is either accurate or appropriate.

Look at that, you were able to prove me wrong, and do something I was not able to do. Thanks for correcting my language; project, product, don't care.   So much of modern day arguing is about _who_ is right, instead of _what_ is write. The documentation is terrible. I don't really care what you think is _appropriate_.  I voiced my concern; just because I didn't explain my complete build environment doesn't mean the documentation shouldn't be more clear.   If someone makes changes, great! If not, whatever.  I got it to work, my own way, without the documentation.  

`There is a big problem in the open-source community with maintainers becoming burnt out because of the sometimes entitled/aggressive/abusive language and behavior from users of free software. Please do not contribute to that.`

Sure thing, its people like you who make me not want to use a product, project, or whatever pedantic nomenclature you wish. ME stating an opinion, which you are allowed to disagree with, doesn't not make it an entitled opinion. I am ENTITLED to voice my opinion, and you are entitled to disagree.  Documentation and communication can ALWAYS be more clear. That should never be in dispute.

I'm not submitting anything to anyone, no suggestions, no changes, no nothing. I was willing, but at this point you have ruined any chance to get anything from me. I made my frustrations clear. The documentation can be more clear over all.  Period.  We need to be less careful with language: less general, less obfuscation, and less guarded.  I am all for working together, but if you think I am being entitled, abusive, and aggressive, I would hate to see what _kind_ language looks like. 

I'll just uninstall tinygo at this point and never look back.  Please close this issue.

closed 2023-12-16 by JohnRichard