tinygo-org/tinygo #4038: [QUESTION] Is there USB Host support in TinyGo?

shreeve, 2023-12-11:
Is there USB Host support in TinyGo? Or, is this a pretty hairy problem that's not quite been addressed yet?

deadprogram, 2023-12-11:
Not as of yet, but probably mostly because of more people currently wanting to create peripherals, and less wanting to consume them.

Please see https://github.com/tinygo-org/tinygo/tree/dev/src/machine/usb for the current implementation for the peripheral side, which is where host support would probably go as well.

GeoffreyPlitt, 2025-03-21:
I'm interested in this too. Is anyone working on it?

shreeve, 2025-03-21:
I wrote a C based USB Host library for the rp2040/rp2350 which is totally independent from TinyUSB.

It can enumerate devices but I don't have drivers yet, etc. But, the main thing is that I was able to figure out the low-level details of how the rp2040/rp2350's USB Host mode works. As far as I know, it's the only non-TinyUSB library to do this.

The code is at: https://github.com/shreeve/picousb

It would need to be ported to TinyGo, but the "hard part" of figuring out the low level specifics is complete, the core data structures are there, enumeration works, and some basic driver support is there.

At a minimum, it could be very helpful for someone who is fluent in TinyGo to be able to create new USB Host support for at least the rp2040/rp2350.

GeoffreyPlitt, 2025-03-21:
That is really exciting! I'm hoping someone with more C knowledge can help with the TinyGo side, but I can help verify. I have Pi Pico (both 2040 and 2350) boards, and an `Adafruit Feather RP2040 with USB Type A Host`.