tinygo-org/tinygo #4040: reflect: setting an interface value

radkomih, 2023-12-12:
I have encountered an issue and was able to create a small example to reproduce it. 
When running the code with standard Go, it prints out the value '13', however, with TinyGo, it prints out '35008' instead.

**Env:** `tinygo version 0.30.0 darwin/amd64 (using go version go1.21.0 and LLVM version 16.0.1`

`tinygo run -target=wasi main.go`

```go
package main

import (
	"reflect"
)

func main() {
	var value interface{} = uint16(0)

	// get the pointer to the interface value
	inPtr := reflect.ValueOf(&value)

	// dereference to get the actual value (an interface)
	inElem := inPtr.Elem()

	// create a new value of the same concrete type
	uint16Type := reflect.TypeOf(uint16(0))
	newUint16Value := reflect.New(uint16Type).Elem()
	newUint16Value.Set(reflect.ValueOf(uint16(13)))

	// set the new value to the interface
	inElem.Set(newUint16Value)

	println(value.(uint16))
}
```

dgryski, 2023-12-13:
Thanks for the bug report.  I'll see what I can do.

dgryski, 2023-12-13:
Can you please try https://github.com/tinygo-org/tinygo/pull/4041 and see if that fixes your problem?  (It solves your test case, but I want to make sure it works on your real code base.)

radkomih, 2024-02-05:
@dgryski it works for my case
thank you ❤️

PR #4041 (2024-01-31): reflect: move indirect values into interface when setting interfaces
Fixes #4040
commit 5b8127fd (2024-01-31): reflect: move indirect values into interface when setting interfaces

Fixes #4040
closed 2024-02-26 by deadprogram