tinygo-org/tinygo #4042: export and extern not working

jclab-joseph, 2023-12-14:
```
package main

import (
	"fmt"
)

func main() {
	fmt.Println("ptr: ", ptr)
}

//export test
func test(a int) {
	fmt.Println("tets ", a)
}

//extwrn test
var ptr uintptr
```

Result:
```
function declaration may only have a unique !dbg attachment
ptr @test.4
error: verification error after compiling package command-line-arguments
```

fgsch, 2023-12-14:
What version of tinygo are you using and how are you compiling this?

jclab-joseph, 2023-12-14:
@fgsch tinygo version 0.31.0-dev linux/amd64 (using go version go1.21.3 and LLVM version 15.0.7)

fgsch, 2023-12-14:
Right, but what command line are you using to compile this?

jclab-joseph, 2023-12-14:
No other options. `tinygo run main.go`.
The same goes for play.tinygo.org.

fgsch, 2023-12-17:
Works here with:

tinygo version 0.30.0 darwin/amd64 (using go version go1.21.5 and LLVM version 16.0.1)

under macos Sonoma. 

Can you repro with 0.30.0?

jclab-joseph, 2023-12-18:
There was a typo error in the code.
Could you please try again?

Same error in `tinygo version 0.30.0 linux/amd64 (using go version go1.21.3 and LLVM version 16.0.1)`.

```
$ tinygo version
tinygo version 0.30.0 linux/amd64 (using go version go1.21.3 and LLVM version 16.0.1)
$ tinygo run main.go
function declaration may only have a unique !dbg attachment
ptr @test.2
error: verification error after compiling package command-line-arguments

```

fgsch, 2023-12-18:
Ah, yes. I can reproduce it 👍

fgsch, 2023-12-18:
I'm a bit confused though, you are exporting different symbols with the same name. This is not going to work.

jclab-joseph, 2023-12-18:
When I saw that a C function was imported with go:extern, I thought it would be possible since it has the same role.

Any ideas on how to get an exported function pointer without a syscall import dependency?

deadprogram, 2025-02-22:
Closing due to age/inactivity. Please reopen if needed.

closed 2025-02-22 by deadprogram

aykevl, 2025-02-24:
You can use `reflect.ValueOf(fn).Pointer()`.