tinygo-org/tinygo #4044: tinygo uses wrong go version

Jnoack331, 2023-12-15:
I Just followed the instruction for ubuntu to install tinygo. But instead of a supported Go version, tinygo outputs the following wrong go version:

`tinygo version 0.30.0 linux/amd64 (using go version go1.16.2 and LLVM version 16.0.1)`

so i cant build my project as it requires newer version:
`error: requires go version 1.18 through 1.21, got go1.16`

Is there some quirk i need to follow to fix this?

Jnoack331, 2023-12-15:
I tried changing the go environment to the required go version but that did not work:

```
GOPATH=/home/noack/go/ GOROOT=/home/noack/go/go1.20.12/ tinygo version                                                                                                                                                                                                                                                                                                                                                
tinygo version 0.30.0 linux/amd64 (using go version go1.16.2 and LLVM version 16.0.1)
```

Jnoack331, 2023-12-15:
I fixed it. I uninstalled to default go version from the ubuntu lts version by executing:
`sudo apt remove golang`

This removed the old go version from the PATH. Then i added the new go version, installed via jetbrains goland, to the path by adding the following to my `~/.zshrc` and opening a new terminal window:
`PATH="/home/noack/go/go1.20.12/bin:$PATH"`

This needs to be done because tinygo seems to select the go version via PATH instead of GOROOT.

I hope this is helpful to someone in the same situation :smile:

closed 2023-12-15 by Jnoack331

aykevl, 2023-12-16:
> This needs to be done because tinygo seems to select the go version via PATH instead of GOROOT.

Correct, we just look up the `go` binary. This means that all you need to do to make sure TinyGo is working, is to make sure the `go version` command outputs the right version.