tinygo-org/tinygo #4050: error while importing time: package sync/atomic is not in std

mahdi-khanzadi, 2023-12-18:
I recieve the `error while importing time: package sync/atomic is not in std` error on running the below command:

```
tinygo flash -target pico main.go
```

my code is: 

```go
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.High()
		time.Sleep(time.Second)
		led.Low()
		time.Sleep(time.Second)
	}
}
```

the version of tinygo and Golang I use: 

```
tinygo version 0.30.0 darwin/amd64 (using go version go1.21.2 and LLVM version 16.0.1)

go version go1.21.2 darwin/arm64
```

upils, 2023-12-22:
I have the same issue, with the following command:

```
tinygo build -target=d1mini .
```

I use these versions:

```
tinygo version 0.30.0 windows/amd64 (using go version go1.21.5 and LLVM version 16.0.1)

go version go1.21.5 windows/amd64
```

I tried various go version and it breaks from go 1.21.3 and up, which is a bit weird because AFAIK this release only contains a fix for a CVE (see https://github.com/golang/go/issues?q=milestone%3AGo1.21.3+label%3ACherryPickApproved). And this is also weird because this is working with go 1.21.2 (which is the problematic version for @mahdi-khanzadi).

@mahdi-khanzadi maybe you could for now try to downgrade your go version if possible.

TinHead, 2024-03-26:
I've hit the same issue while trying to setup IDE integration.
It appears that setting GOROOT tp the cached GOROOT (~/.cache/tinygo/goroot ...) causes another cached goroot to appear which does not have the atomic symlink :/ Not sure if I am making sense here, see below .. 
1. clear all cache
2. Run `tinygo info pico`  in my case I get: 
```
LLVM triple:       thumbv6m-unknown-unknown-eabi
GOOS:              linux
GOARCH:            arm
build tags:        cortexm baremetal linux arm rp2040 rp pico tinygo purego math_big_pure_go gc.conservative scheduler.tasks serial.usb
garbage collector: conservative
scheduler:         tasks
cached GOROOT:     /home/razvan/.cache/tinygo/goroot-53b85c802f0ceb56ba098612a829adb2c6931e0f1be8bb881704c4bb0864b379

```
4. Check the cache dir:
```
ls ~/.cache/tinygo/
goroot-53b85c802f0ceb56ba098612a829adb2c6931e0f1be8bb881704c4bb0864b379/
```
5. atomic symlink is in src/sync:
```
ls -al ~/.cache/tinygo/goroot-53b85c802f0ceb56ba098612a829adb2c6931e0f1be8bb881704c4bb0864b379/src/sync/
total 64
drwxr-xr-x 1 razvan users 326 Mar 26 16:25 ./
drwxr-xr-x 1 razvan users 934 Mar 26 16:25 ../
lrwxrwxrwx 1 razvan users  76 Mar 26 16:25 atomic -> /gnu/store/6varfd53pqgylqgklcqhm5rv3aawb5ci-go-1.21.5/lib/go/src/sync/atomic
```
6. Set GOROOT accordingly: 
```
export GOROOT= /home/razvan/.cache/tinygo/goroot-53b85c802f0ceb56ba098612a829adb2c6931e0f1be8bb881704c4bb0864b379
```
8. Now it gets weird the ide finds machine, etc but tinygo fails to build:
```
tinygo build .
../../../.cache/tinygo/goroot-360e87b7a3f2eb84c6404c5c66f68da06b44a86f34fbe5bc6196781daca9be5e/src/runtime/cond.go:7:2: package sync/atomic is not in std (/home/razvan/.cache/tinygo/goroot-360e87b7a3f2eb84c6404c5c66f68da06b44a86f34fbe5bc6196781daca9be5e/src/sync/atomic)
```
9. New goroot was created without the atomic symlink
```
ls -al ~/.cache/tinygo/goroot-360e87b7a3f2eb84c6404c5c66f68da06b44a86f34fbe5bc6196781daca9be5e/src/sync/
total 60
drwxr-xr-x 1 razvan users 314 Mar 26 16:32 ./
drwxr-xr-x 1 razvan users 934 Mar 26 16:32 ../
lrwxrwxrwx 1 razvan users  78 Mar 26 16:32 cond.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/cond.go
lrwxrwxrwx 1 razvan users  83 Mar 26 16:32 cond_test.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/cond_test.go
lrwxrwxrwx 1 razvan users  77 Mar 26 16:32 doc.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/doc.go
lrwxrwxrwx 1 razvan users  77 Mar 26 16:32 map.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/map.go
lrwxrwxrwx 1 razvan users  82 Mar 26 16:32 map_test.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/map_test.go
lrwxrwxrwx 1 razvan users  79 Mar 26 16:32 mutex.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/mutex.go
lrwxrwxrwx 1 razvan users  84 Mar 26 16:32 mutex_test.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/mutex_test.go
lrwxrwxrwx 1 razvan users  82 Mar 26 16:32 oncefunc.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/oncefunc.go
lrwxrwxrwx 1 razvan users  87 Mar 26 16:32 oncefunc_test.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/oncefunc_test.go
lrwxrwxrwx 1 razvan users  78 Mar 26 16:32 once.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/once.go
lrwxrwxrwx 1 razvan users  83 Mar 26 16:32 once_test.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/once_test.go
lrwxrwxrwx 1 razvan users  78 Mar 26 16:32 pool.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/pool.go
lrwxrwxrwx 1 razvan users  83 Mar 26 16:32 pool_test.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/pool_test.go
lrwxrwxrwx 1 razvan users  83 Mar 26 16:32 waitgroup.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/waitgroup.go
lrwxrwxrwx 1 razvan users  88 Mar 26 16:32 waitgroup_test.go -> /gnu/store/0ivwgksq41yxrrgz8a60bq55zxjsgjyp-tinygo-bin-0.31.2/src/sync/waitgroup_test.go
```
10. if I add the symlink manually in there everything works as expected 

To add extra context I'm on guix hence the weird /gnu/store stuff but I don't think that's the cause, it used to work before 0.30
So the question is  why is there a new cached goroot created instead of using the existing one?

deadprogram, 2024-12-06:
Seems like this is a local config issue? Can this issue now be closed?

deadprogram, 2024-12-06:
This is the same issue due to changing the local config as #4450 so now closing. Thanks everyone!

closed 2024-12-06 by deadprogram