tinygo-org/tinygo #4058: USB-CDC issue on USB2 port

caizikun, 2023-12-28:
Here's the code:

package main

import (
	"fmt"
	"strings"

	"machine"
)

func main() {
	err := run()
	if err != nil {
		fmt.Printf("%s\n", err.Error())
	}
	select {}
}

func run() error {

	ser := machine.Serial

	server := CommandServer{

		Messages: []string{
			"hello world",
			"こんにちは 世界",
		},
	}

	i := 0
	input := [256]byte{}
	for {
		if ser.Buffered() > 0 {
			data, _ := ser.ReadByte()

			switch data {
			case 10, 13: // CR or LF
				if i == 0 {
					continue
				}

				ret, err := server.RunCommand(string(input[:i]))
				if err != nil {
					return err
				}
				fmt.Print(ret)
				i = 0
			default:
				input[i] = data
				i++
			}
		}
	}

	return nil
}

type CommandServer struct {
	Messages []string
}

func (s *CommandServer) RunCommand(command string) (string, error) {
	if command == "" {
		return "ok\r\n", nil
	}

	spl := strings.Split(command, " ")
	switch spl[0] {

	case "msg":

		return fmt.Sprintf("%s\r\nok\r\n"), nil
	default:
	}

	return "ok\r\n", nil
}

Clion 2023.2.3 with Win10 LTSC, Raspberry pico.
After flashing pico, plug into USB2 port with Win 10, Win10 cannot recognize pico and cannot install any USB driver, also no VID found. 
But with the same PC and Win10, plug Pico into USB3 port, Pico works normally and USB driver can be installed automated. 

Anybody can reproduce this issue or know the root cause?
Thank you.

deadprogram, 2025-03-16:
I think this is related to recent fixes for rp2040 USB after flashing, so I will tag to close after next release.

deadprogram, 2025-03-19:
Now closing with 0.37.0 release. Thanks!

closed 2025-03-19 by deadprogram