tinygo-org/tinygo #4063: tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 2712 bytes

scraly, 2024-01-02:
In my GBA app, when I wan to display some font, I have an issue.

```
	tinyfont.DrawChar(&display, &notoemoji.NotoEmojiRegular20pt, 5, 140, 'B', green)
```

Error:
```
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 2712 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 2757 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 2795 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 2814 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 2851 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 2887 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 2928 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 2963 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 2999 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3033 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3071 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3107 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3164 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3220 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3245 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3284 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3340 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3378 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3447 bytes
tinygo:ld.lld: error: section '.data' will not fit in region 'iwram': overflowed by 3491 bytes
tinygo:ld.lld: error: too many errors emitted, stopping now (use --error-limit=0 to see all errors)
failed to run tool: ld.lld
error: failed to link /var/folders/lq/xp6s4vbn13s5vj_kq3cch50w0000gn/T/tinygo3415233328/main: exit status 1
```

Do you know why?

scraly, 2024-01-03:
It's what I thought, I have the problem a lot :(

Do you know if it exists a way to "fix" this behavior?

Putting the fonts locally should reduce it for instance?

aykevl, 2024-01-03:
> Putting the fonts locally should reduce it for instance?

What do you mean?

One thing you could do is use a smaller font size, to reduce the impact.

@conejoninja perhaps you can take a look? I believe it's relatively easy to reduce this issue by changing `Glyph.Bitmaps` to a string (strings are stored in flash while byte slices are typically stored in RAM).

scraly, 2024-01-04:
Yes smaller font size reduce a little bit the impact but the RAM is too quickly impacted :).