tinygo-org/tinygo #4080: Tinygo build. But lots of package undefined.

msxfXF, 2024-01-10:
## Version

` tinygo version 0.30.0 darwin/amd64 (using go version go1.21.4 and LLVM version 16.0.1) `

## Source code
```
package main

func main() {
	println("hello world!")
}

```
## Build
`tinygo build -target cortex-m0`

## Output

```
# runtime
/opt/homebrew/Cellar/tinygo/0.30.0/src/runtime/scheduler.go:30:21: undefined: timeUnit
/opt/homebrew/Cellar/tinygo/0.30.0/src/runtime/scheduler.go:84:42: undefined: timeUnit
/opt/homebrew/Cellar/tinygo/0.30.0/src/runtime/time.go:11:33: undefined: timeUnit
/opt/homebrew/Cellar/tinygo/0.30.0/src/runtime/baremetal.go:59:2: undefined: putchar
...
/opt/homebrew/Cellar/tinygo/0.30.0/src/runtime/time.go:12:9: undefined: nanosecondsToTicks
```

But when I change target to *arduino*. Compile normally.
I want to know what steps I overlooked?
Or is the file missing?

deadprogram, 2024-01-13:
Hello @msxfXF the target `-target cortex-m0` is not specific enough to build. You need to use the specific target that matches your *actual board*. For example:

`tinygo build -target arduino-nano33 ./path/to/your/code`

The target named `cortex-m0` is used as a dependency for boards that use a processor based on `m0` to avoid duplication in the `.json` files that describe the various targets.

Hope that helps explain.

msxfXF, 2024-01-13:
This has greatly helped me, thank you very much!

 

 
 
 
 
&mdash;

aykevl, 2024-02-25:
Looks like the issue has been solved, so closing.

closed 2024-02-25 by aykevl