tinygo-org/tinygo #4084: Slower than JavaScript when using WebAssembly

guiferpa, 2024-01-12:
### Context
Currently I'm studying about wasm and for archive this goal it'll build a minimalist image editor.

**Source code repository address**: https://github.com/guiferpa/tinygo-wasm

### Problem
Developing this challenge I'm considering have a performance measuring for each function that apply the filter into image using wasm and js. Given that I figure out that wasm's slower than js. 

- **WebAssembly layer/Go code**: https://github.com/guiferpa/tinygo-wasm/blob/main/main.go#L8
- **JavaScript code**: https://github.com/guiferpa/tinygo-wasm/blob/main/main.js#L1

### Consideration
I'm using `unsafe` package for develop wasm function but I hope that's not the problem.

### Mensuration
[image]

soypat, 2024-01-13:
Went ahead and tried my hand at optimizing the Go filter on amd64 and managed a ~25% increase in speed, maybe it transfers to WASM?

### changed code
The principle applied for optimizing is _slow divide avoidance_. Usually dividing by a constant which is not a power-of-two is one of the slower operations on modern CPUs. We can avoid 3 divides per loop iteration by accumulating into one uint16 and then dividing the result.

```go
func applyBlackAndWhiteFilter(p *uint, size int) {
	up := uintptr(unsafe.Pointer(p))

	i := 0
	for i < size {
		ap0 := (*byte)(unsafe.Pointer(up + uintptr(i+0)))
		ap1 := (*byte)(unsafe.Pointer(up + uintptr(i+1)))
		ap2 := (*byte)(unsafe.Pointer(up + uintptr(i+2)))

		filter := byte((uint16(*ap0) + uint16(*ap1) + uint16(*ap2)) / 3)

		*ap0 = filter
		*ap1 = filter
		*ap2 = filter

		i += 3
	}
}
```

### Results
```go
goos: linux
goarch: amd64
pkg: github.com/soypat/seqs/local
cpu: 12th Gen Intel(R) Core(TM) i5-12400F
BenchmarkBWFilterCanon-12         213970              5056 ns/op
BenchmarkBWFilter-12              289238              3805 ns/op
PASS
ok      github.com/soypat/seqs/local    2.284s
```

guiferpa, 2024-01-15:
The problem it was always me 😅. I was getting elapsed time by wrong way. By the way many thanks for your time and help.

closed 2024-01-15 by guiferpa