tinygo-org/tinygo #4089: dfu_version unexpected in manifest for pca10059?

aeijdenberg, 2024-01-17:
Hi, I've been trying to get the example flashing light working with `tinygo` and the `nRF52840` USB Dongle.

When I run `tinygo flash -target=pca10059 ./main.go` I get the following output:

```
Traceback (most recent call last):
  File "nordicsemi/__main__.py", line 1555, in <module>
  File "click/core.py", line 1137, in __call__
  File "click/core.py", line 1062, in main
  File "click/core.py", line 1668, in invoke
  File "click/core.py", line 1668, in invoke
  File "click/core.py", line 1404, in invoke
  File "click/core.py", line 763, in invoke
  File "nordicsemi/__main__.py", line 1032, in usb_serial
  File "nordicsemi/__main__.py", line 982, in do_serial
  File "nordicsemi/dfu/dfu.py", line 69, in __init__
  File "nordicsemi/dfu/package.py", line 671, in unpack_package
  File "nordicsemi/dfu/manifest.py", line 200, in from_json
TypeError: Manifest.__init__() got an unexpected keyword argument 'dfu_version'
[85994] Failed to execute script '__main__' due to unhandled exception!
```

I'm running this version of `nrfutil` downloaded from <https://www.nordicsemi.com/Products/Development-tools/nrf-util>:

```
% nrfutil --version                   
nrfutil 7.6.0 (92ce3a0 2023-08-15)
commit-hash: 92ce3a0c1cc7e203326d85919d826009510dc631
commit-date: 2023-08-15
host: aarch64-apple-darwin
build-timestamp: 2023-08-23T08:22:36.822784Z
classification: nrf-external
```

and have installed the `nrf5sdk-tools` package:

```
% nrfutil list    
Command        Version  Description
nrf5sdk-tools  1.0.1    nRF5 SDK tools that were available in nRF Util 6
   dfu                  
   keys                 
   pkg                  
   settings             
   zigbee               

Found 1 installed command(s)
```

I noticed the `manifest.json` inside of the package ZIP file generated by Go includes the following content:

```json
{
    "manifest": {
        "application": {
            "bin_file": "application.bin",
            "dat_file": "application.dat",
            "init_packet_data": {
                "application_version": 4294967295,
                "device_revision": 65535,
                "device_type": 82,
                "firmware_crc16": 10853,
                "softdevice_req": [
                    65534
                ]
            }
        },
        "dfu_version": 0.5
    }
}
```

whereas if I generate the package ZIP with the command that replaces, i.e. `nrfutil pkg generate --hw-version 52 --sd-req 0x0 --application application.bin --application-version 1 package.zip` then it generates much simpler `manifest.json`:

```json
{
    "manifest": {
        "application": {
            "bin_file": "application.bin",
            "dat_file": "application.dat"
        }
    }
}
```

Note this does not include the `dfu_version` attribute that `nrfutil` complained about.

That ZIP then works and I've managed to flash my device for the first time!

It's my first time working in this space, happy to help try to fix this with a PR, but wanted to open the issue first in case I'm advertently using incorrect tooling.

The code that generates the `manifest.json` looks to be here https://github.com/tinygo-org/tinygo/blob/731532cd2b6353b60b443343b51296ec0fafae09/builder/nrfutil.go#L15-L24

CC @deadprogram @aykevl who I think made the last change in this space?

aeijdenberg, 2024-01-17:
I reverted https://github.com/tinygo-org/tinygo/commit/6435f62dcc34dd64ac7e5d57532e260a34d11c3e and built from source and verified that does resolve the issue in my environment.

deadprogram, 2024-12-16:
Hello @aeijdenberg just checking to see if this is still an issue for you?

rednerd, 2025-01-17:
I'm encountering this problem with Tinygo 0.35 both using the .deb release and when built from source.   My device is a PCA10059 with the Open DFU bootloader.

When I look at the source for https://github.com/tinygo-org/tinygo/blob/release/builder/nrfutil.go it seems to be generating maninfests for the Nordic Secure Bootloader which won't work with the Open DFU bootloader.

Looking at the docs for this board  and the machine package, I don't seen anything that specifies what bootloader type is required.

https://tinygo.org/docs/reference/microcontrollers/pca10059/
https://tinygo.org/docs/reference/microcontrollers/machine/pca10059/

I'm not real familiar with the Nordic ecosystem yet.   Can someone confirm this?

kdochoa, 2025-02-20:
Also happening to me. @aeijdenberg solution worked

aeijdenberg, 2025-02-20:
> Hello [@aeijdenberg](https://github.com/aeijdenberg) just checking to see if this is still an issue for you?

Sorry for the late reply. And I'm not able to check if it's still an issue - I was playing with this a year ago but don't have the dev environment setup anymore.

From recent comments it looks like others still have the same or similar issue.

commit e5caa99b (2025-02-22): feature: ability to generate nrf firmware manifest for Open DFU format using nrf_open_dfu build tag. Intended to fix #4089