tinygo-org/tinygo #4094: releases for Darwin arm64 for Mac m1 and m2

gedw99, 2024-01-19:
https://github.com/tinygo-org/tinygo/releases/tag/v0.30.0 does not have Darwin arm builds.

gedw99, 2024-02-16:
am back on working with tinygo and was wondering if my need for a release is likely ?

I know brew, etc work, but having the binary in releases makes it work with other CI tooling.
For example aqua, which watches GitHub repo releases to allow using tinygo across many machines.

aykevl, 2024-02-16:
We will have darwin/arm64 release binaries in the coming release (see https://github.com/tinygo-org/tinygo/pull/4113).

Because GitHub Actions doesn't have Windows ARM64 runners, it would be difficult for us to provide windows/arm64 release binaries. You can however build them yourself: I've used TinyGo on Windows ARM64 and it seems to work fine.

If there's enough demand, I could provide a binary that I build in a VM on my laptop but I'd rather not because it's much more complicated and less transparent than a binary built in CI.

aykevl, 2024-02-16:
It seems there is progress on ARM hosted runners, so perhaps we'll eventually have Windows on ARM hosted runners:
https://github.com/github/roadmap/issues/836

gedw99, 2024-02-16:
thanks for the follow up @aykevl 

Darwin arm64 binaries are my first priority. Github CI has a runner for it.

aykevl, 2024-09-21:
> Following up on windows arm64 is possible ?

Once GitHub Actions support it, yes.
The announcements on the ARM based hosted runners say nothing about the OS. I expect them to make Linux available, I don't know whether they'll make Windows ARM available at the same time. But we'll see.

I don't plan on running a whole VM on the MacOS ARM runners (if that's even possible), but I guess if someone puts in the effort it might be worth merging. It just seems like a lot of work for something that will hopefully soon not be needed anymore (and with x86 emulation it should already be possible to run TinyGo on Windows ARM, even if it's a bit slower).