tinygo-org/tinygo #4111: DH11 can't compile in arduino

MushiTheMoshi, 2024-02-03:
Hi there,

First of all , thank you for all the amazing work done here, appreciated.

I have an arduino uno which have connected a dht11 humidity sensor, when compiling get the below error.

aykevl, 2024-02-25:
The example imports `fmt`, which is too big to be usable on an AVR. I suggest using a simpler program that just calls `println` instead.

Closing, because this is not a bug.

closed 2024-02-25 by aykevl

MushiTheMoshi, 2024-02-25:
Thank you so much for your help! quick question, how can I troubleshoot this? any tool/command I may use in the future you can kindly suggest me ?

aykevl, 2024-02-25:
In this case it says the flash area is overflowed, which indicates that the binary is too big. Which essentially means you need to reduce the size. There's no specific tool for anything like this I'm afraid.