tinygo-org/tinygo #4114: WASM code that uses `__multi3` produces module that cannot be parsed by `wasm-opt`

flavio, 2024-02-05:
This is a follow up on issue https://github.com/tinygo-org/tinygo/issues/3501. The issue has been fixed with tinygo 0.29.0.
However, the [code linked as reproducer](https://github.com/flavio/tinygo-wasi-symbol-bug) is still broken. This time the build fails during the optimization phase, when running `wasm-opt`.

The compilation fails with the following error:
```console
[parse exception: invalid function index (at 0:33795)]
```

This happens when building with the following Tinygo docker images:

* `tinygo/tinygo:0.29.0`
* `tinygo/tinygo:0.30.0`

Both images ship with `wasm-opt` version 114. I've tried to use the latest version of `wasm-opt` (v116), but I got the same error.

## Reproducer

I've updated the code inside of the original reproducer, it can be found inside of [this GitHub repository](https://github.com/flavio/tinygo-wasi-symbol-bug).

flavio, 2024-02-27:
This is happening also with version 0.31.0. I've updated the reproduced to reflect that

aykevl, 2024-03-12:
I found that #3844 was actually incorrect. I have fixed this locally, I'll make a PR soon.

aykevl, 2024-03-12:
Here is the PR that fixes this issue: https://github.com/tinygo-org/tinygo/pull/4188.
I can't confirm the output binary is correct, but at least it compiles.

flavio, 2024-03-13:
@aykevl good news, I've checked out the PR you linked above, built tinygo from scratch and then I successfully built and tested the real code (not the small reproducer). The fix is working :partying_face:

aykevl, 2024-03-13:
@flavio thanks for testing!

closed 2024-06-16 by deadprogram