tinygo-org/tinygo #4124: ESP32 I2C, ADC, PWM Support

cansintartici, 2024-02-12:
Hi all, 
I have just noticed TinyGo and willing to use it for my simple project that I will program an ESP32.
Although in [here ](https://tinygo.org/docs/reference/microcontrollers/esp32-coreboard-v2/ )it is mentioned that I2C, ADC and PWM is not supported yet in ESP32 [this ](https://tinygo.org/docs/reference/microcontrollers/esp32-coreboard-v2/) ticket made me confused which mentions that I2C is supported by TinyGo. 
So my question is, are these protocols not yet supported in TintGo or is it possible to be the wep page is not updated yet?
Thanks for everybody for their efforts in the project.

soypat, 2024-02-13:
ESP32 support is not quite there yet. That said:

- I2C can work via software: https://github.com/tinygo-org/drivers/tree/release/i2csoft
- Depending on the frequency of the PWM, you can modulate it via software as well rather easily leveraging the TinyGo scheduler (using time.Sleep)

If possible I'd suggest using the Raspberry Pi Pico W instead which is near feature complete in TinyGo and supports I2C, ADC and PWM.

cansintartici, 2024-02-13:
Thanks for the response. I will go with pico W

deadprogram, 2024-05-13:
The ESP32 i2c support just landed in `dev` branch https://github.com/tinygo-org/tinygo/pull/4259