tinygo-org/tinygo #4133: Can't flash raspberry pi pico because of missing subpackage device/rp

cswank, 2024-02-18:
When trying to flash a raspberry pi pico I get the following error:

`tinygo flash -target=pico .
../../.cache/tinygo/goroot-67b9bbbc4b1312608c9e71f167ff82793f2d673dc4aa54440a127d983b0a6125/src/machine/board_pico.go:6:2: package device/rp is not in std (/home/craig/.cache/tinygo/goroot-67b9bbbc4b1312608c9e71f167ff82793f2d673dc4aa54440a127d983b0a6125/src/device/rp)` 

This occurs due to the following imports in src/machine:

board_ae_rp2040.go
board_badger2040.go
board_challenger_rp2040.go
board_feather_rp2040.go
board_gopher-badge.go
board_kb2040.go
board_macropad-rp2040.go
board_nano-rp2040.go
board_pico.go
board_qtpy_rp2040.go
board_thingplus_rp2040.go
board_tufty2040.go
board_waveshare-rp2040-zero.go
board_xiao-rp2040.go
machine_rp2040_adc.go
machine_rp2040_clocks.go
machine_rp2040.go
machine_rp2040_gpio.go
machine_rp2040_i2c.go
machine_rp2040_pll.go
machine_rp2040_pwm.go
machine_rp2040_resets.go
machine_rp2040_rng.go
machine_rp2040_rtc.go
machine_rp2040_spi.go
machine_rp2040_sync.go
machine_rp2040_timer.go
machine_rp2040_uart.go
machine_rp2040_usb_fix_usb_device_enumeration.go
machine_rp2040_usb.go
machine_rp2040_watchdog.go
machine_rp2040_xosc.go

aykevl, 2024-02-18:
I guess you built TinyGo manually?
To fix this issue, you need to run `make gen-device`, as described here: https://tinygo.org/docs/guides/build/#additional-requirements

cswank, 2024-02-18:
Whoopsie, yes, that was it.  I missed that.  Thanks a bunch.

closed 2024-02-18 by cswank