tinygo-org/tinygo #4138: tinygo cgo compile error

wopashugeixianshi, 2024-02-20:
**when I use tinygo to compile wasm plugin:**
`tinygo build -o main.wasm  --no-debug -target=wasi main.go`
**compile error. The error log is:**
`# os/user `
`../../../../../usr/local/go/src/os/user/cgo_lookup_cgo.go:14:6: not implemented: build constraints in #cgo line
../../../../../usr/local/go/src/os/user/cgo_lookup_cgo.go:18:10: fatal: 'pwd.h' file not found
../../../../../usr/local/go/src/os/user/getgrouplist_unix.go:12:10: fatal: 'grp.h' file not found`
**Once I disabled the cgo compile parameters:**
`$ CGO_ENABLED=0 tinygo build -o main.wasm  --no-debug -target=wasi main.go`
**The error log is:**
`# golang.org/x/sys/unix`
`../../../../../usr/local/Go/pkg/mod/golang.org/x/sys@v0.16.0/unix/syscall_linux.go:1901:64: undefined: syscall.Rlimit
../../../../../usr/local/Go/pkg/mod/golang.org/x/sys@v0.16.0/unix/syscall_linux.go:1906:50: undefined: syscall.Rlimit
../../../../../usr/local/Go/pkg/mod/golang.org/x/sys@v0.16.0/unix/syscall_linux.go:1906:79: undefined: syscall.Rlimit
../../../../../usr/local/Go/pkg/mod/golang.org/x/sys@v0.16.0/unix/syscall_unix.go:605:17: undefined: syscall.Setrlimit
../../../../../usr/local/Go/pkg/mod/golang.org/x/sys@v0.16.0/unix/syscall_unix.go:605:47: undefined: syscall.Rlimit`
**tinygo version**
`tinygo version 0.30.0 linux/amd64 (using go version go1.21.5 and LLVM version 16.0.1)`

Is there any good idea?
I have read some relevant links but still cannot resolve them。
references:
> https://github.com/tinygo-org/tinygo/issues/3044
> https://tinygo.org/docs/reference/lang-support/stdlib/

aykevl, 2024-02-20:
This will be partially fixed by #4056, which implements a custom os/user package for TinyGo.
Can you try again using the dev branch?

wopashugeixianshi, 2024-02-21:
> This will be partially fixed by #4056, which implements a custom os/user package for TinyGo. Can you try again using the dev branch?

I used an image named tinygo/tinygo-dev: latest，but encountered an error
>```
>/go/pkg/mod/github.com/open-policy-agent/opa@v0.61.0/plugins/rest/rest.go:16:2: package net/http/httputil is not in std (/root/.cache/tinygo/goroot-c8af6ada495b364e84af7f22189e7827eb79e4877cf342ff1c7dc04db3114f34/src/net/http/httputil)
>/go/pkg/mod/golang.org/x/net@v0.20.0/http2/transport.go:26:2: package net/http/httptrace is not in std (/root/.cache/tinygo/goroot-c8af6ada495b364e84af7f22189e7827eb79e4877cf342ff1c7dc04db3114f34/src/net/http/httptrace)
>/go/pkg/mod/github.com/docker/docker@v25.0.1+incompatible/registry/search_session.go:10:2: package net/http/cookiejar is not in std (/root/.cache/tinygo/goroot-c8af6ada495b364e84af7f22189e7827eb79e4877cf342ff1c7dc04db3114f34/src/net/http/cookiejar)
>```
>tinygo version
>```
>tinygo version 0.31.0-dev-5baee9a linux/amd64 (using go version go1.21.7 and LLVM version 17.0.1)
>```

deadprogram, 2024-02-28:
Completed with `v0.31.0` so now closing. Thank you!

closed 2024-02-28 by deadprogram